FreeCAD/FreeCAD #6159: use multiple mesh in analysises

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4560

* **Reporter:** Jee-Bee
* **Date submitted:** 2/11/2021
* **FreeCAD version:** 0.2
* **Category:** Feature
* **Status:** acknowledged
* **Tags:** 

# Original report text

ATM it is not possible to solve analysises with multiple meshes

# Additional information

https://forum.freecadweb.org/viewtopic.php?p=476836

# Steps to reproduce

-

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 2/15/2021

# Discussion from Mantis ticket

### Comment by berndhahnebach 2021-02-15 05:54
BTW: it is possible to solve tie constraints with gaps in the surface. All bodies have to be meshed in the same huge mesh. 

But still ... The use of multiple meshes would be an great enhancement.

SYehChen, 2022-11-17:
I create an example model for test. Here is the model and a pdf file for descripting the details of the model.
There are 4 parts in the model, glue together by ConnstarintTied.
. FreeCAD FEM complaint that "multiple mesh is not supported".

[Shaft_FEM_test.zip](https://github.com/FreeCAD/FreeCAD/files/10029044/Shaft_FEM_test.zip)

luzpaz, 2022-11-17:
@SYehChen can you bump the forum thread as well to let them know you've uploaded an example? TIA!

SYehChen, 2022-11-29:
I already did.

luzpaz, 2022-11-29:
@donovaly can you take a look if you have a moment?

donovaly, 2022-11-30:
> @donovaly can you take a look if you have a moment?

Not the next weeks. Sorry.

luzpaz, 2022-12-01:
@berndhahnebach can you take a look at this?

maxwxyz, 2024-05-07:
@FEA-eng FYI, this regularly comes up

FEA-eng, 2024-05-07:
> @FEA-eng FYI, this regularly comes up

Yes, I often have to explain it on the forum and I described the current workarounds here: https://wiki.freecad.org/FEM_Geometry_Preparation_and_Meshing#Assembly_geometries