FreeCAD/FreeCAD #6160: OpenCamLib (OCL) raises an error on incorrect tool attributes and causes FreeCAD segfault

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4561

* **Reporter:** sliptonic
* **Date submitted:** 2/11/2021
* **FreeCAD version:** 0.19
* **Category:** Bug
* **Status:** new
* **Tags:** 

# Original report text

https://forum.freecadweb.org/viewtopic.php?f=15&t=55240

The sample file requires the legacy tools to be enabled but appears to be related to passing incorrectly defined tools to OCL.

FreeCAD: /home/brad/opencamlib/src/cutters/ballcutter.cpp:41: ocl::BallCutter::BallCutter(double, double): Assertion `length>0.0' failed.
Aborted (core dumped)

Tool type: Engraver
FreeCAD: /home/brad/opencamlib/src/cutters/conecutter.cpp:46: ocl::ConeCutter::ConeCutter(double, double, double): Assertion `center_height > 0.0' failed.
Aborted (core dumped

# Additional information

May not affect all platforms.
Known to crash on Mint and Fedora

We don't plan to fix legacy tools to prevent bad user data but handling of the OCL error should be fixed

# Steps to reproduce

Open the cone crasher file.  Change the surface operation to use one of the other tools and recompute.

# FreeCAD Info

```
OS: Linux Mint 20 (i3/i3)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.23964 +1 (Git)
Build type: Unknown
Branch: (HEAD detached at upstream/pr/4383)
Hash: 9dce167073d90460478efb0d38056e4fcb116e6e
Python version: 3.8.5
Qt version: 5.12.8
Coin version: 4.0.0
OCC version: 7.3.0
Locale: English/United States (en_US)

```

# Other bug information

* **Priority:** normal
* **Severity:** crash
* **Category:** Bug
* **Updated:** 8/29/2021

# Discussion from Mantis ticket

### Comment by fremen 2021-02-11 17:23
C++ assert()  will cause a core dump unless NBEBUG is set. That may need doing for release builds.

---

### Comment by fremen 2021-02-15 09:52
https://github.com/FreeCAD/FreeCAD/pull/4433

---

### Comment by StefanBruens 2021-02-22 21:37
The pull request has been merged, so this can be closed, correct?

---

### Comment by fremen 2021-02-23 07:59
The PR I provided fixed two specific errors in FC which were triggering coredumps.  The question remains, is this error trap outcome desirable in a stable release or should it be trapped/handled differently. 

If there is a future error condition is discovered, is it desirable that the user gets faced with a core dump? 

This was clearly intended as debugging outcome in OCL and there is a variable to turn it off. Since OLC is currently compiled outside of FC and to all intents and purposes has no active developer this cannot be fixed from FC. 

One option would be to fork OCL into FC tree and detect the error condition in a more contorlled manner or submit a PR to OLC if that is preferable. 

In short the PR fixed a local bug triggering an instance of the problem, it does not fix the problem.

---

### Comment by Kunda1 2021-08-29 18:29
c:FreeCAD:ab186aa05ac75:

---

### Comment by Kunda1 2021-08-29 18:31
sliptonic when you get a chance, can you weigh in on this ?

luzpaz, 2022-02-11:
Forum thread has been [bumped](https://forum.freecadweb.org/viewtopic.php?p=570558#p570558).

sliptonic, 2022-03-23:
Closing.  Besides the trapping from the noted PR,  we're also deprecating legacy tools and have better control with the toolbits so it's easier to avoid the problem.

closed 2022-03-23 by sliptonic