FreeCAD/FreeCAD #6161: Excessive vertices generated for curved pocket on drawing projection which hangs application

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4564

* **Reporter:** mattbaker.digital
* **Date submitted:** 2/12/2021
* **FreeCAD version:** 0.19
* **Category:** Bug
* **Status:** feedback
* **Tags:** 

# Original report text

Generating an XZ projection of the following part in the TechDraw WB results in an excessive number of vertices for the curved cut-out and blocks the UI thread (unresponsive).
!(https://i.imgur.com/9Xve7yM.gif)

Original forum post: [https://forum.freecadweb.org/viewtopic.php?f=3&t=55489&p=477314#p477264](https://forum.freecadweb.org/viewtopic.php?f=3&t=55489&p=477314#p477264)

# Steps to reproduce

Open the attached model (Tube.FCStd)
Open the drawing page and set the Front projection direction to X0, Y-1, Z0.000000
FreeCAD hangs while generating the view and has an excessive number of vertices when complete.
!(https://i.imgur.com/AKSEiAq.png)

Set the projection direction to X0, Y-1, Z0.000001
The view is now missing lines.
!(https://i.imgur.com/sGG8KdQ.png)

Set the projection direction to X0, Y-1, Z0.000100
The view is now almost perfect.
!(https://i.imgur.com/XFqSa4F.png)

# FreeCAD Info

```
OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.23756 (Git)
Build type: Release
Branch: master
Hash: 9c6e9184930a52b165a0b7274e3a45d1006bfe67
Python version: 3.8.6
Qt version: 5.12.5
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United Kingdom (en_GB)
```

# Other bug information

* **Priority:** normal
* **Severity:** major
* **Category:** Bug
* **Updated:** 1/24/2022

# Discussion from Mantis ticket

### Comment by mattbaker.digital 2021-02-13 00:06
Errors while uploading source files. Uploading again.

---

### Comment by wandererfan 2022-01-02 17:18
is there a connection with 4639?  Overlapping lines not drawn??

---

### Comment by wandererfan 2022-01-10 13:19
The missing short lines seem to be a bug in OCCT.  The number of vertices is appropriate for the projected BSpline.

---

### Comment by Kunda1 2022-01-24 17:45
Can someone test this in the latest (OCC 7.6) ?

grubertechnic, 2023-05-15:
@WandererFan : with your scrubbing algorithm (one turn is enough), the excessive points are vanished. It still need ~3s for the recomputing for this shape, but i do not think that is a real issue. Closing?

```
OS: Debian GNU/Linux 11 (bullseye) (X-Cinnamon/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33184 (Git)
Build type: Release
Branch: master
Hash: c33588be3e9e2bf555b133186a974d7dbd0b48f6
Python 3.9.2, Qt 5.15.2, Coin 4.0.0, Vtk 9.0.1, OCC 7.7.1
Locale: English/United States (en_US)
Installed mods: 
  * QuickMeasure 2022.10.28
  * Assembly4 0.12.6
  * CurvedShapes 1.0.4
  * Curves 0.6.8
```

[image]

WandererFan, 2023-05-15:
closed as completed by edge scrubbing fix c82cf008f3.

closed 2023-05-15 by WandererFan