FreeCAD/FreeCAD #6164: PartDesign Boolean Cut:  attempted translation along X-axis causes unexpected rotation

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4574

* **Reporter:** dg1727
* **Date submitted:** 2/20/2021
* **FreeCAD version:** 0.19
* **Category:** Bug
* **Status:** confirmed
* **Tags:** 

# Original report text

PartDesign Boolean Cut:  attempted translation along X-axis causes unexpected rotation

# Additional information

The same issue also seems to happen (I didn't use exactly the same steps to duplicate it) on v0.18.4, with the same versions of Coin & OpenCASCADE.  

There aren't any errors in the terminal, nor in the FreeCAD logfile.  

# Steps to reproduce

Open FreeCAD.  

Enable the axis cross.  

In the following steps, after each time a solid is made (those steps are marked with a "*"), adjust the view scale & angle (preferably to an oblique view) so you can tell the shape, location, & orientation of what was made.  

In the Part Design workbench, create a Body (the default name is Body).  

Within that Body, create a sketch in the XZ plane.  

Draw a 10mm x 10mm square whose lower-left corner is at (X = 10 mm, Z = 10 mm).  

Exit the sketcher.  

In the tree view, select Body.  In the properties windowpane below the tree view, on the Data tab, enter a value for Placement > Position > x of 50 mm.  

(*) Revolve the sketch, leaving the default parameters, which use the vertical sketch axis as the revolution axis.  This produces a square-cross-section toroid.  The default name is "Revolution."  

Create a new Body (the default name is Body001).  

Within this Body, create a sketch in the XY plane.  

Draw a 10mm-diameter circle whose center is at (X = 65 mm, Y = 15 mm).  

Exit the sketcher.  

(*) Pad the sketch by 30 mm.  (Keep the other parameters at default.)  This makes a cylinder shape which appears to intersect the toroid.  

In the treeview, make Body active.  (Right-click > "Toggle active body") 

Start a Boolean operation.  

In the taskpane for the Boolean, set the Boolean operation to Cut.  

In the taskpane, select "Add body."  Then use the 3D view to select the cylinder shape (Body001).  

Even before you click OK, the toroid (actually the result of the Boolean, a different object than the original toroid) will appear, unexpectedly rotated 90 degrees about the X axis.  

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **Platform:** Linux
* **Updated:** 2/21/2021

# Discussion from Mantis ticket

### Comment by Kunda1 2021-02-21 01:48
Please paste full [About](https://wiki.freecadweb.org/Std_About) info
dg1727 Please first open a forum thread per our forum guidelines. Thank you

---

### Comment by Kunda1 2021-02-21 11:31
dg1727 ping

---

### Comment by dg1727 2021-02-21 12:56
https://forum.freecadweb.org/viewtopic.php?f=8&t=55853

OS: Linux
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19pre from freecad-daily PPA
Build type: Release
Branch: unknown
Hash: ed02d1bbddaa6daeedfe766b2ce33e88f4b22f93
Python version: Same issue on 2 widely different Python versions
Qt version: irrelevant to this issue
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: [redacted]

---

### Comment by dg1727 2021-02-21 13:02
It seems that the issue happens only if the first PartDesign Body (object name "Body" in the procedure above) is a solid of revolution.  (Using a Pad, for instance, doesn't cause the issue.)

---

### Comment by wmayer 2021-02-21 13:09
Confirmed with:

OS: Ubuntu 18.04.5 LTS (XFCE/xubuntu)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24188 (Git)
Build type: Debug
Branch: master
Hash: 385efeeb589dc2b6c3046ace9c9df23a996ced0c
Python version: 3.6.9
Qt version: 5.9.5
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: German/Germany (de_DE)

dholstei, 2023-02-18:
Does anyone have on this yet?

I too, started with a solid of revolution and the boolean rotates one of the objects 90 deg.  Eventually, I changed the plane of the sketch, recalculated the part and was able to do my boolean correctly

luzpaz, 2023-02-18:
Please paste full [About](https://wiki.freecad.org/About) info

luzpaz, 2024-10-27:
Is this issue still relevant ?

wwmayer, 2024-10-27:
Yes.

wwmayer, 2024-10-27:
These are steps to create the geometries:
```
import FreeCAD as App
import Sketcher
import Part

doc = App.newDocument()
body = doc.addObject('PartDesign::Body','Body')
sketch = body.newObject('Sketcher::SketchObject','Sketch')
sketch.AttachmentSupport = (doc.getObject('XZ_Plane'),[''])
sketch.MapMode = 'FlatFace'
doc.recompute()

geoList = []
geoList.append(Part.LineSegment(App.Vector(10.000000, 10.000000, 0.000000), App.Vector(20.000000, 10.000000, 0.000000)))
geoList.append(Part.LineSegment(App.Vector(20.000000, 10.000000, 0.000000), App.Vector(20.000000, 20.000000, 0.000000)))
geoList.append(Part.LineSegment(App.Vector(20.000000, 20.000000, 0.000000), App.Vector(10.000000, 20.000000, 0.000000)))
geoList.append(Part.LineSegment(App.Vector(10.000000, 20.000000, 0.000000), App.Vector(10.000000, 10.000000, 0.000000)))
sketch.addGeometry(geoList, False)
del geoList

constraintList = []
constraintList.append(Sketcher.Constraint('Coincident', 0, 2, 1, 1))
constraintList.append(Sketcher.Constraint('Coincident', 1, 2, 2, 1))
constraintList.append(Sketcher.Constraint('Coincident', 2, 2, 3, 1))
constraintList.append(Sketcher.Constraint('Coincident', 3, 2, 0, 1))
constraintList.append(Sketcher.Constraint('Horizontal', 0))
constraintList.append(Sketcher.Constraint('Horizontal', 2))
constraintList.append(Sketcher.Constraint('Vertical', 1))
constraintList.append(Sketcher.Constraint('Vertical', 3))
sketch.addConstraint(constraintList)
del constraintList
doc.recompute()

body.Placement = App.Placement(App.Vector(50,0,0), App.Rotation(App.Vector(0,0,1),0))

revolve = body.newObject('PartDesign::Revolution','Revolution')
revolve.Profile = (sketch, ['',])
revolve.ReferenceAxis = (sketch,['V_Axis'])
revolve.Angle = 360.0
revolve.Reversed = 1
sketch.Visibility = False
revolve.Angle = 360.000000
revolve.ReferenceAxis = (sketch, ['V_Axis'])
revolve.Midplane = 0
revolve.Reversed = 1
revolve.Type = 0
revolve.UpToFace = None
doc.recompute()

body2 = doc.addObject('PartDesign::Body','Body001')
body2.Label = 'Body001'
sketch2 = body2.newObject('Sketcher::SketchObject','Sketch001')
sketch2.AttachmentSupport = (doc.getObject('XY_Plane001'),[''])
sketch2.MapMode = 'FlatFace'

geoList = []
geoList.append(Part.Circle(App.Vector(46.249985, 27.738132, 0.000000), App.Vector(0.000000, 0.000000, 1.000000), 6.921171))
sketch2.addGeometry(geoList,False)
del geoList

constraintList = []
sketch2.addConstraint(Sketcher.Constraint('Diameter',0,13.842342))
sketch2.setDatum(0,App.Units.Quantity('10.000000 mm'))
sketch2.addConstraint(Sketcher.Constraint('DistanceX',0,3,46.249985))
sketch2.addConstraint(Sketcher.Constraint('DistanceY',0,3,27.738132))
sketch2.setDatum(1,App.Units.Quantity('65.000000 mm'))
sketch2.setDatum(2,App.Units.Quantity('15.000000 mm'))
doc.recompute()

pad = body2.newObject('PartDesign::Pad','Pad')
pad.Profile = (sketch2, ['',])
pad.ReferenceAxis = (sketch2,['N_Axis'])
sketch2.Visibility = False
pad.Length = 30.000000
doc.recompute()
```
The last step i.e. the creation of the boolean cut is not fully scriptable and still must be done manually.

jspiros, 2025-03-14:
I'm running into a problem that seems to be the same as this bug, though in my case it seems related to feature suppression. I am using FreeCAD 1.0.0 on Debian, though I will test on a macOS system momentarily to confirm that it occurs there too.

### What's Happening

I'm trying to use a Fasteners workbench ScrewTap as the tool body to cut a thread into a PartDesign Body with PartDesign Boolean. The PartDesign Body in question is built, in part, using an AdditivePipe.

When I create the Boolean, it works as expected, it cuts and everything is oriented correctly. However, as I want this threaded hole to be optional per parameters, if I set the Suppressed property of that Boolean feature to true, and then back to false, upon unsuppressing, the Boolean causes the entire PartDesign Body's solid to rotate 90 degrees about the X axis (such that what was the "Top" face in the previous feature is now the "Front" face). It still performs the cut, and the cut at least looks correct, but this breaks everything that happens after the Boolean (because nothing is where it's supposed to be anymore).

If I suppress again, it does restore the correct rotation, but any unsuppression of that Boolean feature going forward rotates as before. I can only restore the initial correct cut by deleting and recreating the Boolean feature, and never suppressing it.

### Reproducing and Testing the Bug

Trying to reduce this to the simplest case, I created a test PartDesign Body made from a single AdditivePipe, where the path sketch is on the XY plane and the profile sketch is on the XZ plane, and I used a Part primitive as the Boolean tool body. This is reproducible in that case as well.

It is also reproducible if the PartDesign Body is built using an AdditiveLoft, with the same sketch on the XZ plane as the previous test but a different sketch on the XY plane to act as the other profile.

It is also reproducible if the PartDesign Body is built, using the same XZ plane sketch as the previous two tests, with a Revolution that revolves around the Z axis. **HOWEVER**, it is **not** reproducible if the PartDesign Body is built with a Revolution that revolves around the X or Y axes, using the same XY plane sketch from the AdditiveLoft test.

I also tested with Fuse instead of Cut, and that behaves identically: the same thing where it works fine until it has gone through a suppression cycle. This happens regardless of whether the tool body or the PartDesign Body have adjusted placements. Even if the tool body is non-overlapping and I set it to cut, such that the operation does effectively nothing, it still rotates the body after a suppression cycle, in the cases where it's reproducible.

jspiros, 2025-03-14:
I've never looked into the FreeCAD codebase with any depth, but I decided to poke around anyway to see if anything stood out (so, keep my lack of familiarity in mind, I might not be noticing anything important). But, I thought it was interesting that the suppression code seems to be setting placement for either its own shape or related shapes in two slightly different ways (on lines 127 and 140:

https://github.com/FreeCAD/FreeCAD/blob/9a6a98f824436656cae242ae26b043f78411ebba/src/Mod/PartDesign/App/Feature.cpp#L122-L144

As well, the Boolean feature code fetches a placement value using a "globalPlacement" method which it then applies to the tool shapes, and I'm guessing it's not typical for PartDesign features to pay attention to global placement like this, as it is its own Boolean-specific setting that determines whether that global placement is used, though it seems impossible to set that setting ahead of time, as it's always on by default (per the Gui):

 and it doesn't seem to make a difference for this bug after the Boolean is created), and so maybe there's some weird interaction happening with the previous suppression code?

https://github.com/FreeCAD/FreeCAD/blob/9a6a98f824436656cae242ae26b043f78411ebba/src/Mod/PartDesign/App/FeatureBoolean.cpp#L117

I don't know, but, this did inspire me to use the Python console to examine the Placement of various objects in one of my test cases, a PartDesign Body using a Revolution, followed by a Pad. So, before the Boolean:

```
>>> App.getDocument("Unnamed").getObject("Body").Placement
Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,0)]
>>> App.getDocument("Unnamed").getObject("Revolution").Placement
Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,90)]
>>> App.getDocument("Unnamed").getObject("Pad").Placement
Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,90)]
```

After adding the Boolean, but before suppression cycling (so, when it looks correct and works properly):

```
>>> App.getDocument("Unnamed").getObject("Body").Placement
Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,0)]
>>> App.getDocument("Unnamed").getObject("Revolution").Placement
Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,90)]
>>> App.getDocument("Unnamed").getObject("Pad").Placement
Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,90)]
>>> App.getDocument("Unnamed").getObject("Boolean").Placement
Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,0)]
```

I then add another feature, a pocket, and it's still working fine:

```
>>> App.getDocument('Unnamed').getObject('Body').Placement
Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,0)]
>>> App.getDocument('Unnamed').getObject('Revolution').Placement
Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,90)]
>>> App.getDocument('Unnamed').getObject('Pad').Placement
Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,90)]
>>> App.getDocument('Unnamed').getObject('Boolean').Placement
Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,0)]
>>> App.getDocument('Unnamed').getObject('Pocket').Placement
Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,0)]
```

Next, I suppress the boolean. Here's what I get immediately after suppressing it:

```
>>> App.getDocument('Unnamed').getObject('Body').Placement
Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,0)]
>>> App.getDocument('Unnamed').getObject('Revolution').Placement
Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,90)]
>>> App.getDocument('Unnamed').getObject('Pad').Placement
Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,90)]
>>> App.getDocument('Unnamed').getObject('Boolean').Placement
Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,90)]
>>> App.getDocument('Unnamed').getObject('Pocket').Placement
Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,90)]
```

So, the Boolean has rolled (and the Pocket with it), but it's suppressed, and right now, everything is working fine. The Pocket is still valid, as I used an edge that existed at the Pad to create the Pocket, and which the Boolean did not affect either way. Now, I'll unsuppress the Boolean. This breaks the Pocket, as now it can't find that edge, even though it could with it suppressed, and before it was suppressed. The placements have not changed from the most recent (everything but the body itself shows a 90 degree roll). And, if I hide the Pocket (since it's broken now) and look at the Boolean, the bug has appeared, and the shape is now rotated 90 degrees.

If I create a new Body, and start it with a Pad, instead of a Revolution:

```
>>> App.getDocument('Unnamed').getObject('Body001').Placement
Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,0)]
>>> App.getDocument('Unnamed').getObject('Pad001').Placement
Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,0)]
```

And if I add a Boolean to that one, the Boolean's placement never changes, and remains this way regardless of whether it's suppressed or not:

```
>>> App.getDocument('Unnamed').getObject('Body001').Placement
Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,0)]
>>> App.getDocument('Unnamed').getObject('Pad001').Placement
Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,0)]
>>> App.getDocument('Unnamed').getObject('Boolean').Placement
Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,0)]
```

I hope this makes it obvious what's going on for someone who understands this codebase better than I do. It seems to me that the Boolean is getting its placement from the global placement of the Body, and "resetting" the placement of all features that follow it, but then when it's suppressed/unsuppressed, it's adopting the placement set by the Revolution, which for whatever reason has a roll associated with it. I assume I'd see the same thing with any of the affected features (so, at least AdditivePipe and AdditiveLoft as well).

jspiros, 2025-03-14:
I've never looked into the FreeCAD codebase with any depth, but I decided to poke around anyway to see if anything stood out (so, keep my lack of familiarity in mind, I might not be noticing anything important).

I thought it was interesting that the suppression code seems to be setting placement for either its own shape or related shapes in two slightly different ways (on lines 127 and 140):

https://github.com/FreeCAD/FreeCAD/blob/9a6a98f824436656cae242ae26b043f78411ebba/src/Mod/PartDesign/App/Feature.cpp#L122-L144

Also interesting, the Boolean feature code fetches a placement value using a "globalPlacement" method which it then applies to the tool shapes.

https://github.com/FreeCAD/FreeCAD/blob/9a6a98f824436656cae242ae26b043f78411ebba/src/Mod/PartDesign/App/FeatureBoolean.cpp#L117

Even just as a FreeCAD user I know it's not typical for PartDesign features to pay attention to global placement like this, and it is its own Boolean-specific setting that determines whether that global placement is used. But it seems impossible to set that setting ahead of time, as it's always on by default (per the Gui code):

https://github.com/FreeCAD/FreeCAD/blob/9a6a98f824436656cae242ae26b043f78411ebba/src/Mod/PartDesign/Gui/Command.cpp#L2343

And it doesn't seem to make a difference, for this bug, if that setting is changed after the Boolean is created.

So, I'm thinking maybe there's some interaction happening where the tool bodies having their position set based on the Body's global placement is determining the final feature's shape (post-boolean operation) placement differently from how a feature's placement would normally be determined, but the suppression code is resetting the placement assuming a typical feature, which would be based on the previous feature's placement?

Not sure if I had this right, I was at least inspired to use the Python console to examine the Placement of various objects in one of the test cases, a PartDesign Body using a Revolution, followed by a Pad. So, before the Boolean:

```
>>> App.getDocument("Unnamed").getObject("Body").Placement
Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,0)]
>>> App.getDocument("Unnamed").getObject("Revolution").Placement
Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,90)]
>>> App.getDocument("Unnamed").getObject("Pad").Placement
Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,90)]
```

After adding the Boolean, but before suppression cycling (so, when it looks right and works correctly):

```
>>> App.getDocument("Unnamed").getObject("Body").Placement
Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,0)]
>>> App.getDocument("Unnamed").getObject("Revolution").Placement
Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,90)]
>>> App.getDocument("Unnamed").getObject("Pad").Placement
Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,90)]
>>> App.getDocument("Unnamed").getObject("Boolean").Placement
Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,0)]
```

I then add another feature, a pocket, and it's still working fine:

```
>>> App.getDocument('Unnamed').getObject('Body').Placement
Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,0)]
>>> App.getDocument('Unnamed').getObject('Revolution').Placement
Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,90)]
>>> App.getDocument('Unnamed').getObject('Pad').Placement
Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,90)]
>>> App.getDocument('Unnamed').getObject('Boolean').Placement
Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,0)]
>>> App.getDocument('Unnamed').getObject('Pocket').Placement
Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,0)]
```

Next, I suppress the boolean. Here's what I get immediately after suppressing it:

```
>>> App.getDocument('Unnamed').getObject('Body').Placement
Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,0)]
>>> App.getDocument('Unnamed').getObject('Revolution').Placement
Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,90)]
>>> App.getDocument('Unnamed').getObject('Pad').Placement
Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,90)]
>>> App.getDocument('Unnamed').getObject('Boolean').Placement
Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,90)]
>>> App.getDocument('Unnamed').getObject('Pocket').Placement
Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,90)]
[... 8 lines cut ...]
```
>>> App.getDocument('Unnamed').getObject('Body001').Placement
Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,0)]
>>> App.getDocument('Unnamed').getObject('Pad001').Placement
Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,0)]
```

And if I add a Boolean to that one, the Boolean's placement never changes, and remains this way regardless of whether it's suppressed or not:

```
>>> App.getDocument('Unnamed').getObject('Body001').Placement
Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,0)]
>>> App.getDocument('Unnamed').getObject('Pad001').Placement
Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,0)]
>>> App.getDocument('Unnamed').getObject('Boolean').Placement
Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,0)]
```

So, it seems to me that:

1. The Boolean is getting its initial correct placement from the global placement of the Body, and "resetting" the placement of all features that follow it, but then when it's suppressed/unsuppressed, it's adopting the placement set by the previous feature, and

2. Revolutions, for whatever reason, establish a roll for itself and all following features' placements. (I assume I'd see the same thing with any of the affected features, so, at least AdditivePipe and AdditiveLoft as well.) This isn't noticeable until we hit the Boolean since it seems to assume it's working with a global placement, either due to how it was initialized by the Gui, or just how it's implemented in general (I don't know).

If I have this right, I hope this makes it obvious what's going on for someone who understands this codebase better than I do, and perhaps makes it easier to fix. I suppose I could try to figure out how to fix it myself if I get impatient and have more time, but this is enough from me for now.

FEA-eng, 2026-10-12:
[image]

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 27.1.0dev.20261006 (Git shallow)
Build date: 2026/10/06 21:00:40
Build type: Release
Branch: (HEAD detached at 3424c4106)
Hash: 3424c410627263394f701aeec5b61d7bab7bb3a1
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 474.82/NVIDIA Corporation/Quadro K4000/PCIe/SSE2
```