FreeCAD/FreeCAD #6166: Prefer new-style/functor QObject::connect over string based

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4576

* **Reporter:** StefanBruens
* **Date submitted:** 2/27/2021
* **FreeCAD version:** 
* **Category:** Bug
* **Status:** new
* **Tags:** Qt

# Original report text

New-style or functor based connects allow compile-time checking of signal/slot arguments.

Usage of small lambdas often also allows to get rid of small helper slots/functions.

E.g. (src/Gui/Filedialog.cpp):

```
    connect(this, SIGNAL(filterSelected(const QString&)),
            this, SLOT(onSelectedFilter(const QString&)));
```

becomes
```
    connect(this, &Filedialog::filterSelected,
            this, &Filedialog::onSelectedFilter);
```

# Additional information

https://doc.qt.io/qt-5/signalsandslots-syntaxes.html

https://wiki.qt.io/New_Signal_Slot_Syntax

# FreeCAD Info

```
OS: openSUSE Tumbleweed (KDE//usr/share/xsessions/plasma5)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.Unknown
Build type: Release
Python version: 3.8.7
Qt version: 5.15.2
Coin version: 3.1.3
OCC version: 7.5.1
Locale: English/United States (en_US)

```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **Updated:** 8/27/2021

# Discussion from Mantis ticket

### Comment by wmayer 2021-02-28 07:07
With v0.19 we still support Qt4 (and Py2) which doesn't support the offered method. Now with the started development of v0.20 we will drop Qt4/Py2 support and can start to use the new stuff.

---

### Comment by Kunda1 2021-03-28 11:44
wmayer should we do this before or after Toponaming patch (realthunder is planning to submit a patch by the end of the month, is what he mentioned to me)

---

### Comment by chennes 2021-08-27 02:59
This is not really a bug, more of a "going forward we should code like this" -- while I agree with the sentiment, and my recent PRs have followed this practice, I don't think it's something we want to go back through the whole codebase and replace. We can just update over time as the code gets modified.

luzpaz, 2022-06-07:
@chennes should we consider having a github action that scans new PRs for this pattern and suggests the indicated replacement ?

chennes, 2022-06-07:
I'm totally amenable to it: we should probably have a generic set of checks that run, and that can be one of them. Obviously it's not going to happen until we get 0.20 out the door, though.

luzpaz, 2022-06-07:
@Floriansimmer have you seen a GH action like that int he wild?

luzpaz, 2022-06-07:
Actually, just doing a simple regex string `connect\(this,` search,  I count give or take 4 dozen instances. There are also `disconnect(this` statements.

chennes, 2022-06-07:
It's all-caps SIGNAL and all-caps SLOT that we want to look for and reject -- the initial part of the line is often the same between the two styles.

donovaly, 2022-07-23:
Now that 0.20 is our we can refactor the code to fix this.

luzpaz, 2022-07-24:
@donovaly maybe after toponaming? realthunder is experiencing a lot of annoying cosmetic changes as he works on merging toponaming, see his status update https://forum.freecadweb.org/viewtopic.php?p=611902#p611902

luzpaz, 2023-01-09:
We should do this in conjunction with #8130

donovaly, 2023-01-09:
> We should do this in conjunction with #8130

No, because #8130 would be an easy task while using QObject::connect needs a lot of time.

However, in the meantime we already introduced many QObject::connect and will continue. 
What would help is if you can post here a list of instances where QObject::connect is not yet used.

luzpaz, 2023-01-09:
Instances: https://github.com/FreeCAD/FreeCAD/search?p=3&q=%22this%2C+SIGNAL%22

chennes, 2023-01-10:
Just to port over my comment from #6167 -- this is a high priority item now, it is blocking our migration to Qt6.

StefanBruens, 2023-01-10:
@chennes - how do you come to the conclusion Qt6 no longer supports `char*` based `QObject::connect`? Documentation seems to disagree:
https://doc.qt.io/qt-6/qobject.html#connect

chennes, 2023-01-10:
It's not that they don't work in general, it's that they are broken in various places throughout our codebase, and Qt no longer provides diagnostics. See @wwmayer's post here: https://forum.freecadweb.org/viewtopic.php?p=650866#p650866

StefanBruens, 2023-01-10:
> It's not that they don't work in general, it's that they are broken in various places throughout our codebase, and Qt no longer provides diagnostics. See @wwmayer's post here: https://forum.freecadweb.org/viewtopic.php?p=650866#p650866

I don't see any specific problem mentioned there.

Also, there are definitely warnings, see e.g.:
https://codebrowser.dev/qt6/qtbase/src/corelib/kernel/qobject.cpp.html#2752

donovaly, 2023-01-12:
I found an hour to fix some of the remaining occurrences: #8170 

@chennes , can you please revise and merge if OK?

donovaly, 2023-01-25:
I think meanwhile @wwmayer did all the work and we can close this issue, right?

donovaly, 2023-01-31:
> I think meanwhile @wwmayer did all the work and we can close this issue, right?

I meanwhile changed some more but there are many more to resolve -> we are onto it but it will take months.

wwmayer, 2023-02-03:
> It's not that they don't work in general, it's that they are broken in various places throughout our codebase, and Qt no longer provides diagnostics.

After all only a very few connect() were broken. And for clarification there is no issue with Qt reporting failed connections, it was my bad by not setting a debug build.

Anyway, when searching the code base for SIGNAL (options: case-sensitive, complete words and *.cpp files) then there 55 hits:
* Gui/Application.cpp: must still be done but requires some internal changes
* Gui/DlgActionsImp: it's a bit tricky because the object is known as QWidget and that a pointer to a member function cannot be used
* Gui/PropertyPage.cpp: same here
* Gui/PropertyItem.cpp: this isn't easily possible because the connect() are used inside functions where the slot function is passed as function argument as a string. Since the arguments of the slot function vary it's not possible to use a function pointer
* Part/PropertyEnumAttacherItem: same here
* Web/BrowserView.cpp: it only affects the old webkit binding which isn't available in Qt6 anyway
* others: they are all commented code

Now the next issue is all widget classes with `Q_SLOTS`. Here the slot functions shouldn't be connected with `QMetaObject::connectSlotsByName` either because this is error-prone. Usually a slot method follows the form: `on_OBJECTNAME_SIGNAL`. The connection will fail if the object name or the signal name doesn't match.

luzpaz, 2024-12-13:
@mosfet80 any interest in taking this task on ?