FreeCAD/FreeCAD #6167: Use normalized signatures in QObject::connect when new-style is not possible

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4577

* **Reporter:** StefanBruens
* **Date submitted:** 2/27/2021
* **FreeCAD version:** 
* **Category:** Bug
* **Status:** new
* **Tags:** 

# Original report text

QObject::connect uses normalized signatures when connecting signals to get rid of ambiguities.

Using non-normalized signatures has a runtime overhead (at connect time, not signal emission).

In general, new-style (functor) connects should be preferred, but there may be cases when late binding is preferred (i.e. when the exact type of the target QObject is not known at compile time).

# Additional information

https://marcmutz.wordpress.com/effective-qt/prefer-to-use-normalised-signalslot-signatures/

# FreeCAD Info

```
OS: openSUSE Tumbleweed (KDE//usr/share/xsessions/plasma5)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.Unknown
Build type: Release
Python version: 3.8.7
Qt version: 5.15.2
Coin version: 3.1.3
OCC version: 7.5.1
Locale: English/United States (en_US)

```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **Updated:** 2/27/2021

luzpaz, 2022-06-07:
@StefanBruens was this ever addressed ? (I see no forum thread associated with issue)

luzpaz, 2023-01-10:
Related to #6166 ?

chennes, 2023-01-10:
This is actually no longer relevant: starting with Qt6 we _must_ use the new-style connect calls, we have no choice but to remove all old-style connections.

closed 2023-01-10 by chennes