FreeCAD/FreeCAD #6168: Sketch (diamond shape) collapses on itself and ignores some constraints

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4579

* **Reporter:** garya
* **Date submitted:** 2/28/2021
* **FreeCAD version:** 0.19
* **Category:** Bug
* **Status:** new
* **Tags:** sketcher, Solver

# Original report text

Forum discussion:
  https://forum.freecadweb.org/viewtopic.php?f=3&t=56029&sid=cae9731349d24c2a6929a890bf2208b6

# Steps to reproduce

- load the file Bug_Sketcher_NotRedundant_2_5_simplified.FCStd
- edit sketch Sk_Diamond
- set length of Edge2 to 0.4mm
-> The diamond shape collapses on itself, the top two legs folding down onto the bottom
   This is a violation of the L2_nrm_L3 constraint
-> solver reports 1 DOF
-> Report View shows 
  Updating geometry: Error build geometry(2): Both points are equal
  Invalid solution from SQP(augmented system) solver
- close sketcher
-> an error is reported for the sketch. 

# FreeCAD Info

```
OS: FreeBSD 11.4-RELEASE-p6 (XFCE/xfce)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.b2ca86d8d
Build type: Release
Python version: 3.7.9
Qt version: 5.15.2
Coin version: 4.0.0
OCC version: 7.5.0
Locale: C/Default (C)

```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **Updated:** 2/28/2021

maxwxyz, 2024-12-21:
@AjinkyaDahale is this still relevant?

AjinkyaDahale, 2025-07-18:
Can confirm still happens in the following. No ETA for fixing.

```
OS: Ubuntu Core 22 (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.1.0dev.42596 (Git) Snap 1580
Build date: 2025/07/18 04:00:38
Build type: Release
Branch: main
Hash: c968effe26b199864e6009c03a1f54d22cb89e3b
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 9.1.0, boost 1_74, Eigen3 3.4.0, PySide 5.15.11
shiboken 5.15.11, SMESH 7.7.1.0, xerces-c 3.2.3, IfcOpenShell 0.8.2, OCC 7.7.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * OpenTheme 2025.5.20
```