FreeCAD/FreeCAD #6173: Hiding of constraints can be wrong when more than one constraint exists for the same pair of geometric elements.

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4590

* **Reporter:** chrisb
* **Date submitted:** 3/9/2021
* **FreeCAD version:** 0.19
* **Category:** Bug
* **Status:** assigned
* **Tags:** 

# Original report text

When more than one constraint exists for the same pair of geometric elements, then it's only the last one which controls all visibilities in 3D view.
[Forum discussion](https://forum.freecadweb.org/viewtopic.php?f=3&t=56099&p=482701#p482701)

# FreeCAD Info

```
OS: macOS 10.15
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24267 (Git)
Build type: Release
Branch: master
Hash: b2ca86d8d72b636011a73394bf9bcdedb3b109b7
Python version: 3.8.8
Qt version: 5.12.5
Coin version: 4.0.0
OCC version: 7.4.0
Locale: C/Default (C)

```

# Other bug information

* **Priority:** low
* **Severity:** minor
* **Category:** Bug
* **Updated:** 3/9/2021

adrianinsaval, 2022-02-17:
looks like this one is already fixed here
```
OS: Windows 10 (10.0)
Word size of FreeCAD: 64-bit
Version: 0.20.27518 (Git)
Build type: Release
Python version: 3.8.12
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.3
Locale: Spanish/Paraguay (es_PY)
```
from the forum:

> This one is fixed in my branch and will be commited together with the other hiding/showing improvements. Thanks for reporting.

@abdullahtahiriyo can you confirm you pushed a fix for this? (and maybe provide a link)

abdullahtahiriyo, 2022-02-17:
Yes, I fixed that one. I think that was towards the end of last year, maybe October/November. If you cannot find it, I will try to fetch it for you when I have some time...

adrianinsaval, 2022-02-17:
I asked just to leave as a reference here in case you had it but I don't think it's critical to have it. Either way, issue can be closed, please do as I don't have permissions to do it.

closed 2022-02-17 by chennes