FreeCAD/FreeCAD #6174: Solver doesn't report redundancy correctly

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4591

* **Reporter:** chrisb
* **Date submitted:** 3/9/2021
* **FreeCAD version:** 0.19
* **Category:** Bug
* **Status:** assigned
* **Tags:** 

# Original report text

When multiple redundancies exist, the solver reports only partially redundant constraints and not fully redundant ones.
[Forum Discussion](https://forum.freecadweb.org/viewtopic.php?f=3&t=55948)

# FreeCAD Info

```
OS: macOS 10.15
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24267 (Git)
Build type: Release
Branch: master
Hash: b2ca86d8d72b636011a73394bf9bcdedb3b109b7
Python version: 3.8.8
Qt version: 5.12.5
Coin version: 4.0.0
OCC version: 7.4.0
Locale: C/Default (C)

```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **Updated:** 3/9/2021

luzpaz, 2023-04-15:
@AjinkyaDahale is this ticket still relevant ?

abdullahtahiriyo, 2023-05-02:
The issue persists. 

If a constraint is fully redundant should not be presented as partially redundant.

```
Output with current master:
16:01:11  EigenSparseQR, Threads: 1, Vectorization: On, Pivot Threshold: 1e-13, Params: 44, Constr: 46, Rank: 44
16:01:11  Analysing groups of constraints of special interest:
[1 7 7 0 0 0 0 ]
[10 2 2 8 8 9 9 10 7 7 0 0 0 0 0 0 0 0 ]
16:01:11  Chosen redundants: [7 ]
16:01:11  DL: tolg: 1e-80, tolx: 1e-80, tolf: 1e-10, convergence: 1e-10, dogLegGaussStep: FullPivLU, xsize: 44, csize: 45, maxIter: 105
16:01:11  DL: stopcode: 1, Success
16:01:11  Sketcher::RedundantSolving-DogLeg-
16:01:11  Sketcher Redundant solving: 1 redundants
16:01:11  (Partially) Redundant, Group 1, index 9, Tag: 7
16:01:11  (Partially) Redundant, Group 0, index 2, Tag: 7
16:01:11  Sketcher::setUpSketch()-T:0.008
```

PR #9462 (2023-05-03): Sketcher: Fixes wrong redundancy reporting
==========================================

fixes #6174

Problem
=======

The popularity contest heuristic was designed to assume that removing one solver constraint from a redundant group would potentially satisfy it.

This means that for sketcher constraints comprising several solver constraints, the sketcher constraint would never be notified as redundant, but always partially redundant (where the case may be that it is redundant or that it is partially redundant). This happens because after removing one solver constraint, it may happen:

(a) that  [... cut]
commit b626caff (2023-05-03): Sketcher: Fixes wrong redundancy reporting

==========================================

fixes #6174

Problem
=======

The popularity contest heuristic was designed to assume that removing one solver constraint from a redundant group would potentially satisfy it.

This means that for sketcher constraints comprising several solver constraints, the sketcher constraint would never be notified as redundant, but
always partially redundant (where the case may be that it is redundant or that it is partially redundant). This happens because after removing one
solver constraint, it may happen:

(a) that [... cut]