FreeCAD/FreeCAD #6175: [Arch] "Move With Host" property of Arch windows does not work in FC 0.19 /0.20

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4592

* **Reporter:** bitacovir
* **Date submitted:** 3/9/2021
* **FreeCAD version:** 0.2
* **Category:** Bug
* **Status:** new
* **Tags:** 0.19, Arch

# Original report text

In FC 0.19 when I move a Arch Wall with a nested Arch Door or Window, these elements do not move with the arch wall.
As default, "Move With Host" property of windows is setup to "True", but it does not work.
This bug is a regression, because the feature "Move with Host" works normally in FC 0.18.4 for arch windows nested in arch walls.

This bug/regression has been mentioned several times in the forum:
https://forum.freecadweb.org/viewtopic.php?f=23&t=53427
https://forum.freecadweb.org/viewtopic.php?f=23&t=53491
https://forum.freecadweb.org/viewtopic.php?f=23&t=53445

Edit: This bug is also present in FC 0.20.

# Steps to reproduce

In FC0.19 / 0.20
-Open Arch Module
-Create an Arch Wall
-Select the wall and place an Arch window in it. (The property "Move With Host" of the window should be True).
-Select the wall and change or move its position. The wall moves, but the window doesn't.

Same steps in FC 0.18.4 make the wall and windows move simultaneously, as expected.

# FreeCAD Info

```
OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24267 (Git)
Build type: Release
Branch: master
Hash: b2ca86d8d72b636011a73394bf9bcdedb3b109b7
Python version: 3.8.8
Qt version: 5.12.5
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)

OS: Windows 10 (10.0)
Word size of FreeCAD: 64-bit
Version: 0.20.25131 (Git)
Build type: Release
Branch: master
Hash: 7c519689f0d5ea78fb3292be36a857d283c05507
Python version: 3.8.10
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.2
Locale: English/United States (en_US)

```

# Other bug information

* **Priority:** normal
* **Severity:** major
* **Category:** Bug
* **Updated:** 8/6/2021

# Discussion from Mantis ticket

### Comment by matthijskooijman 2021-03-31 19:23
I suspect that the relevant bit of code is `get_movable_children()`. Looking there, it seems to iterate over the OutList of the wall looking for children with "MoveWithHost" set, but that seems wrong to me. Any objects that have the wall set in their "Hosts" property, will (AFAIU) appear in the InList of the wall, not the OutList?

I quickly tried this patch:

```
--- a/src/Mod/Draft/draftutils/groups.py
+++ b/src/Mod/Draft/draftutils/groups.py
@@ -297,8 +297,8 @@ def get_movable_children(objectslist, recursive=True):
     for obj in objectslist:
         # Skips some objects that should never move their children
         if utils.get_type(obj) not in ("Clone", "SectionPlane",
                                        "Facebinder", "BuildingPart"):
-            children = obj.OutList
+            children = obj.InList
             if (hasattr(obj, "Proxy") and obj.Proxy
                     and hasattr(obj.Proxy, "getSiblings")
                     and utils.get_type(obj) != "Window"):
```

Which seems to work to fix this problem. However, it also breaks the copy tool (gives src/App/Expression.cpp:3109: App::ExpressionParser::ExpressionImporter::ExpressionImporter(Base::XMLReader&): Assertion `!_Reader' failed.)

Also, looking at the git history, it seems that this has been OutList since it was first introduced in 2014 ( c:FreeCAD:cb5a80799: ), so this is probably not the right fix.

So I guess the window is supposed to be in the wall's outlist? Should it be in the Additions property? That is what commit c:FreeCAD:cb5a80799: suggests. But if it is in the wall's outlist and the wall is in the window's outlist, then there would be a cycle?

Hm, I guess I do not quite understand yet how all this is supposed to work...

---

### Comment by matthijskooijman 2021-05-22 14:58
See https://forum.freecadweb.org/viewtopic.php?f=23&t=57223 for a lot more detailed investigation about MoveWithHost and its current problems. The above suggested fix certainly is not the right approach :-)

---

### Comment by bitacovir 2021-08-06 16:17
I changed the Severity to mayor as requested here: https://forum.freecadweb.org/viewtopic.php?f=23&t=60679
Also this bugs is present in FC 0.20 so I edit FC version references in the original report.

bitacovir, 2022-08-04:
I think this bug needs attention @yorikvanhavre

yorikvanhavre, 2022-08-06:
it does indeed... i'm in holidays for the next 3 weeks, i'll pick that up when i'm back!

Roy-043, 2023-01-01:
This problem has already been fixed: #5388

closed 2023-01-01 by Roy-043