FreeCAD/FreeCAD #6176: [Arch] Arch Stairs' rails do not recompute in FC 0.19

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4593

* **Reporter:** bitacovir
* **Date submitted:** 3/9/2021
* **FreeCAD version:** 0.19
* **Category:** Bug
* **Status:** confirmed
* **Tags:** 0.19, Arch

# Original report text

when you create a Arch Stairs object, rails components produce some errors and they cannot recompute in FC 0.19
Message of Report panel:
```
21:06:45  0 <App> Document.cpp(3527): Unnamed#Rail still touched after recompute
21:06:45  1.8e-05 <App> Document.cpp(3527): Unnamed#Rail001 still touched after recompute
```

I found this affect the performance of other commands in Draft WB make them slower: when I draw a line, before to display the geometry in the 3Dwindow, the program try to recompute and fails, generating the error messages. The only solution I found is to delete the rails.
This bug is a regression compared with Arch Stair in FC 0.18.4 where you can create a stair and no errors are produced.

Forum discussion: https://forum.freecadweb.org/viewtopic.php?p=485978#p485978

# Steps to reproduce

-New file
-Open Arch module
-Create a Stair with Arch Stairs (click on Stair icon)
-The new stair is generated but two error messages are shown in the report panel.
-If you try to recompute, it will fail.

# FreeCAD Info

```
OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24267 (Git)
Build type: Release
Branch: master
Hash: b2ca86d8d72b636011a73394bf9bcdedb3b109b7
Python version: 3.8.8
Qt version: 5.12.5
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)

```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **OS: Windows 10**
* **Platform:** windows
* **Updated:** 8/20/2021

# Discussion from Mantis ticket

### Comment by Kunda1 2021-03-22 16:34
confirmed on the forum. Marking target as 0.20

---

### Comment by Kunda1 2021-08-20 14:29
paullee is there anyone that can help you with this ticket?

---

### Comment by paullee 2021-08-20 15:29
Thanks, I am not sure.  Maybe I will have a look when there is a gap.

luzpaz, 2022-09-13:
CC @paullee0

Roy-043, 2023-02-13:
Fixed with #8444.