FreeCAD/FreeCAD #6177: Grid is transparent in respect to the window

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4594

* **Reporter:** dudo
* **Date submitted:** 3/10/2021
* **FreeCAD version:** 0.2
* **Category:** Bug
* **Status:** new
* **Tags:** Coin3d

# Original report text

Grid lines, instead of being filled with color, are filled with whatever is "below" the FreeCAD window, eg. desktop wallper, other windows, etc.

https://forum.freecadweb.org/viewtopic.php?f=3&t=56509

https://streamable.com/4xedi5

https://i.imgur.com/dVKiDaB.png

# Additional information

This is using fully up to date Arch Linux, using git version of FreeCAD, running under wayland / sway

# FreeCAD Info

```
OS: Arch Linux
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.20.24407 (Git)
Build type: Release
Branch: master
Hash: 9f937f765d9c811535018110618056883b258837
Python version: 3.9.2
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US)

```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **OS: Linux **
* **Platform:** Arch Linux
* **Updated:** 5/23/2021

# Discussion from Mantis ticket

### Comment by dudo 2021-03-11 01:16
It seems to be related to blending, like every mesh that's rendered with some kind of transparency shows the bug in action

https://streamable.com/0qavx3

(Additionally, note the z-fighting issues, unrelated I guess)

---

### Comment by dudo 2021-03-11 17:34
On the same system under Xorg it works all right, no transparency issues, so this could be wayland specific

---

### Comment by Kunda1 2021-03-11 19:47
What graphics card/display driver are you using? Have you read the [known issues](https://forum.freecadweb.org/viewtopic.php?f=3&t=56509&p=486617#p486415) as chrisb mentions in the forum thread?

---

### Comment by dudo 2021-03-12 01:18
I've read the known issues but couldn't find any that would describe this behavior. The driver is Intel/MESA.

```
~> glxinfo | grep OpenGL
OpenGL vendor string: Intel Open Source Technology Center
OpenGL renderer string: Mesa DRI Intel(R) HD Graphics 4000 (IVB GT2)
OpenGL core profile version string: 4.2 (Core Profile) Mesa 20.3.4
OpenGL core profile shading language version string: 4.20
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 3.0 Mesa 20.3.4
OpenGL shading language version string: 1.30
OpenGL context flags: (none)
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 3.0 Mesa 20.3.4
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.00
OpenGL ES profile extensions:
~> lspci | grep VGA
00:02.0 VGA compatible controller: Intel Corporation 3rd Gen Core processor Graphics Controller (rev 09)
```

---

### Comment by dannixon 2021-03-29 17:02
I am also seeing this issue using FreeCAD 0.19.
It happens with both Weston and wlroots based compositors and not at all under Xorg/Xwayland.
I'll also add that setting a solid background color also causes the entire render area to be affected, not just transparent areas.

I do notice the following error which is only seen when running under Wayland:
`Coin warning in glxglue_isdirect(): Couldn't get current GLX context.`

---

### Comment by skewty 2021-05-23 14:43
This also affects Fedora 34 Gnome so Platform field should perhaps be adjusted accordingly.
What I see in Fedora 34 is the part background color, when set to solid is completely transparent.  Oddly this does not occur on the gradient setting.
A work-around for this is thus to set both gradient colors to be the same (looks like a solid color but isn't transparent).

I am also getting odd Qt menu placement issues in Wayland.  For example:

*  File menu opens to the left outside window bounds.
*  Edit menu opens to the top outside window bounds.

Additionally the splash screen is not centered when it opens for me.

Could these all be related to a screen size / boundary call (upstream in Qt most likely) not working correctly in Wayland?

---

### Comment by skewty 2021-05-23 14:55
```
[scott@fedora ~]$ glxinfo | grep OpenGL
OpenGL vendor string: Intel
OpenGL renderer string: Mesa Intel(R) UHD Graphics (CML GT2)
OpenGL core profile version string: 4.6 (Core Profile) Mesa 21.1.1
OpenGL core profile shading language version string: 4.60
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 4.6 (Compatibility Profile) Mesa 21.1.1
OpenGL shading language version string: 4.60
OpenGL context flags: (none)
OpenGL profile mask: compatibility profile
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 3.2 Mesa 21.1.1
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
OpenGL ES profile extensions:
[scott@fedora ~]$ lspci | grep VGA
00:02.0 VGA compatible controller: Intel Corporation CometLake-U GT2 [UHD Graphics] (rev 02)
[scott@fedora ~]$ 
```

luzpaz, 2022-04-21:
@PLChris This is most likely Graphic Driver related right?

luzpaz, 2022-04-28:
@wwmayer is this that old graphic driver issue?

norpol, 2022-06-25:
I'm having the same issue on Swaywm/Wayland, but only when I'm installing the package from the archlinux packages. Downloading the latest appimage from the FreeCAD website and then launching the executable doesn't have that bug.

DMNinja, 2023-06-11:
I can also confirm this issue still exists

This is the output of FreeCAD Info

```
OS: Artix Linux (Hyprland)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: makepkg
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.11.3, Qt 5.15.9, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * BIM 2021.12.0
  * boltsfc 2022.11.5
  * ProDarkThemePreferencePack 1.0.0
  * Help 1.0.3
  * ExtremeProDark 2.6.4
```

[image]
Sorry in advance for my bug reporting inexperience. Please, feel free to ask for any futher info you might need

yorikvanhavre, 2023-06-12:
This all seems to happen on wayland... It would be interesting to know what graphics driver you guys are using?

luzpaz, 2023-06-13:
@yorikvanhavre an aside, maybe it's worth having About info post graphics driver info ?

yorikvanhavre, 2023-06-14:
Yes, certainly... but I don't know if it's very easy to get that info on all platforms

adrianinsaval, 2023-06-14:
> Downloading the latest appimage from the FreeCAD website and then launching the executable doesn't have that bug.

because that runs on xwayland

this rather seems like rather a bug in wlroots since it seems to be specific to Sway/Hyprland
I never experienced something like this in KDE's wayland session and I have an nvidia card (and nvidia is supposed to be the one with poor wayland support). Can one of you try it on kde or gnome wayland session?

luzpaz, 2023-10-28:
bumping for discussion

luzpaz, 2023-12-01:
@DMNinja can you still reproduce on latest stable 0.21.2  ?  (https://github.com/FreeCAD/FreeCAD/releases/tag/untagged-0d1019172178853f7266)

flaviut, 2024-05-10:
Related issue in another app with a proposed solution from a wayland developer: https://github.com/swaywm/sway/issues/6549

AjinkyaDahale, 2024-05-22:
Not able to reproduce after opening a simple model in "BIM" and setting their transparency.

[image]

```
OS: Ubuntu Core 22 (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37442 (Git) Snap 1027
Build type: Release
Branch: main
Hash: 6b5f829ca20229a1bd6e670d9f9d13c80862210f
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United States (en_US)
```

flaviut, 2024-05-22:
The freecad appimage doesn't support wayland & runs in X11 mode, so I wouldn't be surprised if the snap is also running in X11 mode. I can reproduce this 100% in wayland mode (by using a native binary) by just opening a blank sketch:

[image]

```
OS: Arch Linux (sway)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: makepkg
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.12.3, Qt 6.7.0, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * kicadStepUpMod 11.1.4
  * Glass
```

flaviut, 2024-12-08:
I can't reproduce this anymore on d37115c692e66d3c5aa821eaec57e7b3dd0f1cd4. It seems the new sketch UI has changed, but the transparency is working property for bodies too.

```
OS: Arch Linux (sway/wayland)
Architecture: x86_64
Version: 1.1.0dev.39512 (Git)
Build type: Release
Branch: main
Hash: d37115c692e66d3c5aa821eaec57e7b3dd0f1cd4
Python 3.12.7, Qt 5.15.16, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: ProDark.qss/unset/Qt default
```

[image]

norpol, 2024-12-08:
Just tested it too @flaviut and it doesn't show up immediately either. I'm not sure if there is a way to get the grid to show up though to verify that it's fully resolved. I'm on [Freecad 1.0.0 on NixOS unstable](https://github.com/NixOS/nixpkgs/blob/nixos-unstable/pkgs/by-name/fr/freecad/package.nix#L182).

flaviut, 2025-01-05:
Well this is weird, I'm now able to reproduce this. But it seems to only happen when the window is a certain width.

Narrow, no problem:

[image]

Wide, and you can see the desktop background:

[image]

This seems to be related to a specific number of pixels, not the aspect ratio, since it doesn't come up if I make the window shorter:

[image]

```
OS: Arch Linux (sway/wayland)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.13.0, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * OpenTheme 2024.9.1
  * sheetmetal 0.6.13
  * kicadStepUpMod 11.4.4
```

adrianinsaval, 2025-08-02:
this might be a qt bug: https://bugreports.qt.io/browse/QTBUG-132197

FEA-eng, 2026-08-19:
See [this](https://github.com/FreeCAD/FreeCAD/issues/6177#issuecomment-3869673049)