FreeCAD/FreeCAD #6178: [Arch] Arch External Reference does not load windows nested in walls

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4595

* **Reporter:** bitacovir
* **Date submitted:** 3/13/2021
* **FreeCAD version:** 0.19
* **Category:** Bug
* **Status:** new
* **Tags:** 

# Original report text

Arch External Reference does not display windows nested in walls, inside an Arch BuildingPart, in FC 0.19
This is a regression, because the External Reference works normally in FC 0.18.4

Forum discussions where this bug is reported:
https://forum.freecadweb.org/viewtopic.php?f=23&t=56467
https://forum.freecadweb.org/viewtopic.php?f=23&t=56566

# Steps to reproduce

-Open Arch Module.
-Create an Arch wall
-Place an Arch window in the wall
-Put the wall in an Arch BuildingPart
-Save the file.
-Open a new file.
-Create an Arch Reference object.
-In the reference panel select the previous file and the Arch BuildingPart object.
-The reference object only shows in the 3Dwindow the wall with the hole, without the window.

# FreeCAD Info

```
OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24267 (Git)
Build type: Release
Branch: master
Hash: b2ca86d8d72b636011a73394bf9bcdedb3b109b7
Python version: 3.8.8
Qt version: 5.12.5
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)

```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **OS: Windows 10**
* **Platform:** Windows
* **Updated:** 3/13/2021

donovaly, 2022-12-01:
Many thanks Roy, I backported this also for FC 0.20.2.

I plan to make the 0.20.2 release on Sunday. Do yo have anything in the pipeline that could be merged/backported?

closed 2022-12-01 by donovaly

Roy-043, 2022-12-02:
The only PR I have currently planned is an improvement of the Arch unit tests (related to this issue). But that is not relevant for V0.20. So in short: No.

PR #7907 (2022-12-01): Arch: Fix missing Windows in Arch Reference of BuildingPart
Fixes #6178.

- [x]  Your Pull Request meets the requirements outlined in section 5 of [CONTRIBUTING.md](https://github.com/FreeCAD/FreeCAD/blob/master/CONTRIBUTING.md) for a Valid PR

---