FreeCAD/FreeCAD #6179: UserAppData dir on macOS is wrong.

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4596

* **Reporter:** pilhuhn
* **Date submitted:** 3/14/2021
* **FreeCAD version:** 0.2
* **Category:** Bug
* **Status:** new
* **Tags:** #pending-PR, macOS

# Original report text

See https://forum.freecadweb.org/viewtopic.php?f=22&t=7104

# Steps to reproduce

See 1st entry in the above forurm thread.

# FreeCAD Info

```
OS: macOS 10.15
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.20.24434 (Git)
Build type: Release
Branch: master
Hash: 8ec3766b00d672c15ce050d6a29a5d76f5602153
Python version: 3.9.0
Qt version: 5.15.2
Coin version: 4.0.0
OCC version: 7.5.0
Locale: German/Germany (de_DE)

```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **Updated:** 3/27/2021

# Discussion from Mantis ticket

### Comment by Kunda1 2021-03-15 21:49
p:FreeCAD:4620:

---

### Comment by pilhuhn 2021-03-16 06:22
See also https://github.com/FreeCAD/FreeCAD/pull/4623 which is related.

luzpaz, 2022-04-11:
@PLChris @chennes @ipatch 
Do y'all mind testing to see if this issue still persists?

ipatch, 2022-04-11:
compiled a recent commit the other day, on my big sur VM

```
OS: macOS 10.15
Word size of FreeCAD: 64-bit
Version: 0.20.28638 (Git)
Build type: Release
Branch: ipatch.builds.master.head.mac.app
Hash: dc3986ee1083e6ad0a8fcc11b00ebec6f8966638
Python 3.9.12, Qt 5.15.3, Coin 4.0.0, OCC 7.6.1
Locale: English/United States (en_US)
Installed mods: 
  * Plot
  * CfdOF
  * Defeaturing
  * CfdOF.bak 1.9.6
  * A2plus 0.4.55b
  * Manipulator 1.4.3
  * fasteners 0.3.40
  * Curves 0.4.1
```

running the commands mentioned in the linked forum thread in the python console

```python
>>> HomeMod = FreeCAD.ConfigGet("UserAppData")+"Mod"
>>> HomeMod = os.path.realpath(HomeMod)
>>> print(HomeMod)
/Users/capin/Library/Application Support/FreeCAD/Mod
```

<details>

<summary> 📸 screenshot </summary>

[image]

</details>

ipatch, 2022-04-11:
and for what it's worth i still see files being written in `$HOME/Library/Preferences/FreeCAD`

```
╭─capin at rogue in /Users/capin/Library/Preferences/FreeCAD
╰─λ l                                                                                                                                        0 (0.000s)
total 264K
drwxrwxr-x   10 capin staff 320 Mar 22 11:53 ./
drwx------+ 716 capin staff 23K Apr 11 12:16 ../
drwxrwxr-x    3 capin staff  96 Nov  8 21:07 AddonManager/
drwxrwxr-x    8 capin staff 256 Mar 22 15:06 Macro/
drwxrwxr-x    4 capin staff 128 Mar 22 11:57 Mod/
drwx------   10 capin staff 320 Nov  3 17:09 webdatapersistent/
-rw-rw-r--    1 capin staff 923 Mar 22 12:01 link.system.cfg
-rw-rw-r--    1 capin staff 90K Mar 22 12:01 link.user.cfg
-rw-rw-r--    1 capin staff 922 Apr 11 12:08 system.cfg
-rw-rw-r--    1 capin staff 82K Apr 11 12:08 user.cfg
```

ipatch, 2022-04-11:
so i just renamed dir `FreeCAD` within the `$HOME/Library/Preferences` to `FreeCADbkp` relaunched FreeCAD.app and sure enough the "launch scrips" recreated the `$HOME/Library/Preferences/FreeCAD` directory.

```
╭─capin at rogue in /Users/capin/Library/Preferences
╰─λ l                                                                                                                                        0 (0.028s)
total 14M
drwx------+ 717 capin staff  23K Apr 11 12:24  ./
drwx------+  99 capin staff 3.1K Apr 10 20:08  ../
drwxr-xr-x    3 capin staff   96 Dec  1  2018  .wacom/
drwxrwxr-x    7 capin staff  224 Oct  5  2021  Adobe/
drwxrwxr-x    3 capin staff   96 May 12  2020  AndroidSDKManager/
drwxrwxr-x    8 capin staff  256 Apr 27  2019  AndroidStudio3.4/
drwx------   84 capin staff 2.7K Apr 11 12:24  ByHost/
drwxrwxr-x    4 capin staff  128 Apr 11 12:24  FreeCAD/
drwxrwxr-x   10 capin staff  320 Mar 22 11:53  FreeCADbkp/
```

luzpaz, 2022-04-24:
CC @pilhuhn

luzpaz, 2022-04-29:
Lets get this macOS stuff fixed before release

donovaly, 2022-05-15:
Sine this bug is older, it is no release blocker. If there won't be a PR to fix it this week, the bug cannot be fixed for FreeCAD 0.20.

ageeye, 2022-06-22:
@luzpaz Please close this ticket. I confirm this [comment](https://github.com/FreeCAD/FreeCAD/pull/4620#issuecomment-1003779129). App.getUserAppDataDir()+"Mod" and the AddOn manager use the postulated path:

/Users/xxx/Library/Application Support/FreeCAD/Mod

luzpaz, 2022-06-22:
Closing ticket. 

Just giving a heads-up to @ipatch and @pilhuhn

closed 2022-06-22 by luzpaz