FreeCAD/FreeCAD #6181: Draft_Wire: Ctrl+Z does not work in FC 0.19

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4603

* **Reporter:** bitacovir
* **Date submitted:** 3/24/2021
* **FreeCAD version:** 0.19
* **Category:** Bug
* **Status:** new
* **Tags:** 0.19, Draft, undo

# Original report text

'Ctrl+Z' does not work inside the Draft_Wire command. Pressing the panel button works, but not the keyboard shortcut.

In FC 0.18.4 'Ctrl+Z' works.

Forum discussion: https://forum.freecadweb.org/viewtopic.php?f=23&t=56805

# Steps to reproduce

Open Draft WB
Click on Draft polyline tool
After two or three clicks try Ctrl+Z. Nothing happens (It should delete last point).

# FreeCAD Info

```
OS: Windows 10 Version 1909
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.20.24451 (Git)
Build type: Release
Branch: master
Hash: 9c8685c20cd70553d85940c2ba93ade69f38171d
Python version: 3.8.6+
Qt version: 5.15.1
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US)

OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24267 (Git)
Build type: Release
Branch: master
Hash: b2ca86d8d72b636011a73394bf9bcdedb3b109b7
Python version: 3.8.8
Qt version: 5.12.5
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)

```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **OS: Windows 10**
* **Platform:** Windows
* **Updated:** 4/1/2021

luzpaz, 2022-02-21:
@bitacovir can you retest when you get a chance?

bitacovir, 2022-02-21:
@luzpaz 
> @bitacovir can you retest when you get a chance?

Yes, still it is there...
OS: Windows 10 (10.0)
Word size of FreeCAD: 64-bit
Version: 0.20.27518 (Git)
Build type: Release
Python version: 3.8.12
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.3
Locale: English/United States (en_US)

luzpaz, 2022-02-21:
Can confirm on
```
OS: Ubuntu Core 20 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.27675 (Git)
Build type: Release
Branch: master
Hash: 4b08dd81422d6801a82c7054b4e1dccdaeb6b616
Python 3.8.10, Qt 5.15.3, Coin 4.0.0, OCC 7.6.1
Locale: English/United States (en_US)
```

luzpaz, 2022-04-14:
@Roy-043 do you want to take a stab at this?

Roy-043, 2022-04-16:
IMO this should become a "normal" incommand shortcut (a single letter without Ctrl). This would then mean that the `ui` file for the preferences also needs to be updated. Since I do not have experience with that, it is better that somebody else fixes this.

Note that AFAICT Ctrl+Z has never worked.

luzpaz, 2022-08-02:
@yorikvanhavre reupping this on your radar, if you don't mind

closed 2022-08-04 by yorikvanhavre

luzpaz, 2022-08-04:
:pray:

commit 5a9201d2 (2022-08-04): Draft: Enabled shortcut for undo segment - fixes #6181

That shortcut had apparently been left out last time the system was recoded..
I used the default shortcut '/' because Ctrl doesn't really work with the
current system, and Z is already taken
commit bfe8b89d (2022-08-05): Draft: Enabled shortcut for undo segment - fixes #6181

That shortcut had apparently been left out last time the system was recoded..
I used the default shortcut '/' because Ctrl doesn't really work with the
current system, and Z is already taken