FreeCAD/FreeCAD #6183: FEM ignore Force / Pressure constraints during *FREQUENCY calculation

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4605

* **Reporter:** Jee-Bee
* **Date submitted:** 3/24/2021
* **FreeCAD version:** 0.2
* **Category:** Bug
* **Status:** new
* **Tags:** calculix, FEM

# Original report text

During a frequency calculation things like masses, forces etc change the result on *FREQUENCY calculation. But these things are ignored in FreeCAD.

# Additional information

See: http://feacluster.com/CalculiX/ccx_2.13/doc/ccx/node8.html
See: https://forum.freecadweb.org/viewtopic.php?f=18&t=20392

See calculix example: http://feacluster.com/CalculiX/ccx_2.13/doc/ccx/input_deck_viewer.php?input_deck=beamf2.inp

# Steps to reproduce

1. Create base part (for example beam
2.  Add fixed constraint on end point
3.  add force constraint on other end in direction of the beam
4.  Run writer
5.  check inp file for output. it isn't there

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Bug
* **Updated:** 3/24/2021

FEA-eng, 2023-12-04:
As explained in the forum thread, in CalculiX, loads influence the frequency analysis only when they are included in the previous static step and perturbation is enabled for the frequency analysis. So this would require support for multiple steps in the FEM workbench which is not possible at the moment.

FEA-eng, 2024-02-18:
@luzpaz or @maxwxyz As explained above, this is currently the expected behavior so I would close this issue.

maxwxyz, 2024-02-19:
Closed as its an expected behavior

closed 2024-02-19 by maxwxyz