FreeCAD/FreeCAD #6187: Export of .obj files inconsistant between Wavefront and Alias Mesh

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4611

* **Reporter:** drmacro
* **Date submitted:** 3/30/2021
* **FreeCAD version:** 0.2
* **Category:** Bug
* **Status:** new
* **Tags:** OBJ

# Original report text

Forum thread: https://forum.freecadweb.org/viewtopic.php?f=3&t=57195

When exporting an assembly:
Wavefront export appears to ignore the deviation property of a feature. Alias mesh uses the deviation value.
Wavefront export adds the feature name to the item in the export, exporting multiple items. Alias mesh exports a single monolithic item

# Steps to reproduce

Create an assembly (I used A2+)
set the deviation value on features to differ values.
Select features export with Wavefront selected.
Select same features, export with Alias Mesh

Import each .obj file in Blender.

# FreeCAD Info

```

OS: Ubuntu 20.04.2 LTS (XFCE/ubuntustudio)
Word size of FreeCAD: 64-bit
Version: 0.20.24640 (Git)
Build type: Unknown
Branch: master
Hash: afa40aa0493309013e2e0dc5199523ef1f4c046c
Python version: 3.8.5
Qt version: 5.12.8
Coin version: 4.0.0
OCC version: 7.5.1
Locale: English/United States (en_US)

```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **OS: Ubuntu 20.04**
* **Platform:** PC
* **Updated:** 3/30/2021

luzpaz, 2022-05-09:
CC @PLChris (just a heads up since you participate on the forum thread)

@yorikvanhavre any idea who could tackle this issue?

adrianinsaval, 2022-05-09:
`obj.ViewObject.AngularDeflection` ? however it's a vale in degrees, not sure what unit is 0.7 in or if it's a percentage

luzpaz, 2022-09-13:
rebumped forum discussion

PR #10989 (2023-10-10): Arch: Fix multiple issues with ImportOBJ.py
Fixes #6187.

Other fixes:
* Made handling of shapes and global placement more consistent.
* Shapes without curves were also triangulated.
* Polygonal faces could have flipped normals and duplicate points.
commit 7029c6f5 (2023-10-16): Arch: Fix multiple issues with ImportOBJ.py

Fixes #6187.

Other fixes:
* Made handling of shapes and global placement more consistent.
* Shapes without curves were also triangulated.
* Polygonal faces could have flipped normals and duplicate points.
commit 930079cd (2023-11-12): Arch: Fix multiple issues with ImportOBJ.py

Fixes #6187.

Other fixes:
* Made handling of shapes and global placement more consistent.
* Shapes without curves were also triangulated.
* Polygonal faces could have flipped normals and duplicate points.
closed 2023-10-10 by yorikvanhavre