FreeCAD/FreeCAD #6190: Request for a "Replacement command" in PartDesign

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4616

* **Reporter:** Pauvres_honteux
* **Date submitted:** 4/5/2021
* **FreeCAD version:** 0.2
* **Category:** Feature
* **Status:** new
* **Tags:** 

# Original report text

See forum discussion: [Feature request - Replacement command](https://forum.freecadweb.org/viewtopic.php?f=8&t=28012)

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 4/5/2021

# Discussion from Mantis ticket

### Comment by Kunda1 2021-04-05 11:04
Changed severity to 'Feature'
Moved ticket from 'Part' to 'PartDesign'

pierreporte, 2024-08-31:
This is a must for wireframe and surface design (see #12699 and #14010).

maxwxyz, 2026-02-01:
Related:

- https://github.com/FreeCAD/FreeCAD/issues/16114