FreeCAD/FreeCAD #6192: Better edge selection for Part Sweep

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4618

* **Reporter:** BloodyRain2k
* **Date submitted:** 4/6/2021
* **FreeCAD version:** 0.19
* **Category:** Feature
* **Status:** new
* **Tags:** 

# Original report text

I've been working a lot with the Part Sweep of the Part Design WB recently and I've started to feel that adding edges of objects shouldn't be as cumbersome as it currently is.

For example is it not possible to preselect the wanted edges and THEN click the "Add Edge" button to add them, nor is it possible to just keep adding them because the button gets "released" after each edge added.
It might not sound like much, but I've had to redo the same Sweep a dozen times while working on my oddly shaped part shown below, because eventually something in the feature order went wrong and I had to delete and later redo the Sweep.

Edit: I also just remembered that the DEL key doesn't work for either the edges nor the profiles used in transition mode.
It's shown in the context menu, but pressing the key doesn't register.

!(https://drive.google.com/uc?id=1kKVFykTwO322jeiS6IgJ6lhzyddIKTie)

On a sidenote: it would be also great if there would be a Compound Filter for just keeping the edges selected when creating it as a path.

# FreeCAD Info

```
OS: Windows 7 Version 6.1 (Build 7601: SP 1)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24276 (Git)
Build type: Release
Branch: releases/FreeCAD-0-19
Hash: a88db11e0a908f6e38f92bfc5187b13ebe470438
Python version: 3.8.6+
Qt version: 5.15.1
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US)

```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **Updated:** 4/6/2021

luzpaz, 2022-02-15:
@BloodyRain2k are you the OP for this ticket?

BloodyRain2k, 2022-02-15:
Seems like, because I remember that weird thing on the screenshot xD

luzpaz, 2022-02-15:
@BloodyRain2k are you planning to resume work on this? If so check to see if there've been changes to the code in master (it's way ahead of 0.19.x)

BloodyRain2k, 2022-02-15:
I'll take a look at it, I wasn't finished with that part anyways and just ignored it for a long while because it was low priority to begin with.

Seems like a good time to try again.

luzpaz, 2022-02-16:
try [v0.20dev](https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds) to test any features

BloodyRain2k, 2022-02-16:
Doesn't seem different in 0.20 (27518) either.
DEL's working an the buttons are still one-adds.

BloodyRain2k, 2022-05-03:
Not sure if there *should* be a change in 0.20.28774, but there still isn't.

Can't add multiple edges in a row nor preselect a couple and THEN press "Add Edge" to add the selected edges.

luzpaz, 2022-05-03:
Please open a forum thread to discuss with the rest of the communtiy

luzpaz, 2023-11-26:
@BloodyRain2k can you kindly upload an example file (zip it and drag/drop into Github textbox) ?

Roy-043, 2024-02-09:
Duplicate: #5755.

closed 2024-02-09 by Roy-043