FreeCAD/FreeCAD #6193: Internal alignment constraints should not be reported as redundant

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4619

* **Reporter:** abdullah
* **Date submitted:** 4/7/2021
* **FreeCAD version:** 0.2
* **Category:** Bug
* **Status:** assigned
* **Tags:** 

# Original report text

An internal alignment constraint can be redundant with other constraints, but it is never the one to be removed.
It may be possible to improve the algorithm suggesting redundant constraints, so that it does not suggest removing internal alignment constraints.

https://github.com/realthunder/FreeCAD/issues/97

# Steps to reproduce

Unfortunately no file at this moment. If somebody comes along with one file, please upload it here.

# FreeCAD Info

```
OS: Ubuntu 20.04.2 LTS (MATE/mate)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.
Build type: Release
Branch: unknown
Hash: e8566f22bbeb0b7204e3c45519d0963e8881100b
Python version: 3.8.5
Qt version: 5.12.8
Coin version: 4.0.0
OCC version: 7.5.1

```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **Updated:** 4/7/2021

AjinkyaDahale, 2023-04-14:
I believe this was fixed recently.

luzpaz, 2023-04-14:
@AjinkyaDahale by whom and would you happen to pinpoint the commit(s)?

AjinkyaDahale, 2023-04-14:
@luzpaz https://github.com/FreeCAD/FreeCAD/commit/aa67808f2412a7ff3d17d138e2445983e9176b18. There may be more.

luzpaz, 2023-04-14:
Closing per AjinkyaDahale

closed 2023-04-14 by luzpaz

AjinkyaDahale, 2023-04-14:
Noticed this was filed by @abdullahtahiriyo himself. Abdullah, please confirm this fixes the issue, and reopen if needed.

abdullahtahiriyo, 2023-04-21:
It should. I am ok with it closed. Thanks.