FreeCAD/FreeCAD #6196: Thickness fails on simple shapes (BRep_API command not done) - regression from 0.19.24267

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4624

* **Reporter:** DocFingo
* **Date submitted:** 4/8/2021
* **FreeCAD version:** 0.19
* **Category:** Bug
* **Status:** new
* **Tags:** 

# Original report text

Thickness operations with **Arc** or **Tangent** join types fail on predominantly concave shapes.
Message is ***BRep_API: command not done***. This happens for any positive Thickness value.

Changing join type to **Intersection** solves the issue, but does not yield the desired part.

This is a potential regression from 0.19.24267 where the issue seemed to be (mostly?) fixed, see reproduction steps below.

A Thickness problem in conjunction with OCC 7.5 was previously reported [here](https://tracker.freecadweb.org/view.php?id=4332) by user AlStu, with forum discussions [here](https://forum.freecadweb.org/viewtopic.php?f=3&t=45642&p=391223&hilit=thickness+limitation#p391223).
[s]However OCC 7.5 is present in both[/s] OCC has been upgraded from 7.3 in 0.19.24267 to 7.5 in 0.19.24276, both show different results.

# Additional information

***This is the previous 0.19.24267 version I used for the comparison:***
OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24267 (Git)
Build type: Release
Branch: master
Hash: b2ca86d8d72b636011a73394bf9bcdedb3b109b7
Python version: 3.6.8
Qt version: 5.12.1
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/Germany (en_DE)

# Steps to reproduce

Part workbench: Make a simple extrusion from a sketch with concave outer shape.
Apply Thickness while selecting the top face. Try different Join types.

To test the potential regression from 0.19.24267, open the attached FCstd file.
Press F5 to recompute the solution.
0.19.24267 -> Thickness is computed successfully except for the sharp-edged shape.
0.19.24276 -> Thickness (Arc/Tangent join type) fails for 3 out of 4 shapes. Deleting and re-applying thickness yields BRep_API: command not done

# FreeCAD Info

```
OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24276 (Git)
Build type: Release
Branch: releases/FreeCAD-0-19
Hash: a88db11e0a908f6e38f92bfc5187b13ebe470438
Python version: 3.8.6+
Qt version: 5.15.1
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/Germany (en_DE)
```

# Other bug information

* **Priority:** normal
* **Severity:** major
* **Category:** Bug
* **OS: Windows 10 Enterprise 10.0.19042**
* **Platform:** Core i5-8350U (x64, 8GB RAM)
* **Updated:** 4/9/2021

BePeMee, 2022-09-04:
I have a similar issue in 0.20.

https://user-images.githubusercontent.com/48593752/188306828-13a6d8fe-c850-455c-a292-94a45438609f.mp4

I have tried to accomplish the same in a commercial CAD package. There it worked, so I guess the geometry of part is not the limiting factor.
Sample file:[CHRhoofdsteun.zip](https://github.com/FreeCAD/FreeCAD/files/9484384/CHRhoofdsteun.zip)
Version info:
[code]
OS: Linux Mint 21 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.20.29177 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.20)
Hash: 68e337670e227889217652ddac593c93b5e8dc94
Python 3.9.13, Qt 5.12.9, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods: 
  * sheetmetal 0.2.56
[/code]

I also tried the thickness feature on a simple cube and that works as expected:
[image](https://i.imgur.com/mxyoV2p.png)

Anthony-Gaudino, 2024-01-07:
Still happens on 0.21.1 when trying to apply thickness to the flat bottom of of this part:  [cant_thickness.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/13853688/cant_thickness.FCStd.zip)

<details>
  <summary>About FreeCAD</summary>
  
  ```
OS: macOS 13.5.2
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * MeshRemodel 1.9.21
  * ThreadProfile 1.89.0
  * Launcher
  * InventorLoader 1.4.0
  * Render 2023.12.28
  * Silk 0.1.5
  * Defeaturing 1.2.1
  * A2plus 0.4.61
  * Manipulator 1.5.6
  * fasteners 0.4.75
  * lattice2 1.0.0
  * CurvedShapes 1.0.5
  * sheetmetal 0.3.14
  * toSketch 1.0.1
  * Curves 0.6.21
  * Assembly4 0.50.6
  * Assembly3 0.12.0
  ```
</details>

JDMils, 2025-03-30:
Is this the same issue:
https://forum.freecad.org/viewtopic.php?t=95930

A simple loft between a rectangle and a circle crashes the Thickness tool. I tested in 0.21.2 onwards and it's been a problem since then so wondering if it will ever get fixed? Is it on the scope? The latest (Windows) build has the same issue.

pieterhijma, 2025-08-31:
I can't retrieve a version of FreeCAD where this does succeed, so although this is a regression in I believe OCCT, it is not something we can compare easily. As such, the regression label is not so helpful here.

Fletch216, 2025-09-10:
I'm a very basic user of Freecad, but I'm learning!  I have been trying to thickness a lofted, padded circle and rectangle to create a transition duct and get the dreaded BRep_API message.  I had a look at anything I could find regarding the error message and ran across one entry (sorry, I forget where it was) that said to increase the thickness and put up with the result.  I tried again (about the 10th time) but used a thickness of 5mm and wham! My duct hollowed out (went and had a beer).  I couldn't use the 5mm wall thickness so I tried going back into the model tree, opened the thickness action and changed the thickness to 2mm, which is what I wanted and wham again, I had the 2mm wall thickness duct.  Another beer to celebrate!

pieterhijma, 2025-09-19:
Minimal model and test code for this regression in OCCT: 

[thickness-regression.zip](https://github.com/user-attachments/files/22430105/thickness-regression.zip)

Reported in https://github.com/Open-Cascade-SAS/OCCT/issues/728.

maxwxyz, 2026-08-15:
closing for no feedback

closed 2026-08-15 by maxwxyz