FreeCAD/FreeCAD #6198: Can't use a linked Sketch for Lofting

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4630

* **Reporter:** BloodyRain2k
* **Date submitted:** 4/11/2021
* **FreeCAD version:** 0.19
* **Category:** Bug
* **Status:** new
* **Tags:** Loft, Sketch

# Original report text

I tried to create something that cuts a hole through a sketch profile into the same sketch but mirrored.
So I've linked the sketch and then flipped it, assuming I can just loft from the original to the linked copy, but it won't do that.

# Steps to reproduce

Create primitive, create sketch, link sketch, try to loft from sketch to linked copy.

Example file included.

# FreeCAD Info

```
OS: Windows 7 Version 6.1 (Build 7601: SP 1)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24276 (Git)
Build type: Release
Branch: releases/FreeCAD-0-19
Hash: a88db11e0a908f6e38f92bfc5187b13ebe470438
Python version: 3.8.6+
Qt version: 5.15.1
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US)

```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **Updated:** 4/11/2021

adrianinsaval, 2022-02-11:
known limitation, Link support is sparse. OP could have used a subshapebinder instead. Also no forum discussion and poor description (Part or Part Desing?). Should be closed IMO

luzpaz, 2025-08-17:
@LemonBoy can you revisit this issue ?

FEA-eng, 2026-02-24:
I attached the file recomputed in 1.2.0, it seems to be working there

[Linked_Loft_1_2.FCStd.zip](https://github.com/user-attachments/files/25513603/Linked_Loft_1_2.FCStd.zip)

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.2.0dev.20260217 (Git shallow)
Build date: 2026/02/17 21:26:14
Build type: Release
Branch: (HEAD detached at 3bad5dabd)
Hash: 3bad5dabde38f1a901fa1475e30f5caee7f9846a
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```

FEA-eng, 2026-09-03:
[image]

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 26.3.0dev.20260902 (Git shallow)
Build date: 2026/09/02 02:30:22
Build type: Release
Branch: (HEAD detached at c68c45227)
Hash: c68c452275ec188f35c02f6b30aba880b3783cf1
Python 3.11.14, Qt 6.8.3, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL) [ OS: Polish/Poland (pl_PL) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 474.82/NVIDIA Corporation/Quadro K4000/PCIe/SSE2
```