FreeCAD/FreeCAD #6199: 3d surface covers model instead of selected faces

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4632

* **Reporter:** monkeydog
* **Date submitted:** 4/13/2021
* **FreeCAD version:** 0.19
* **Category:** Bug
* **Status:** new
* **Tags:** 

# Original report text

When selecting all outside edges of a shape for a single 3D surfacing operation, the entire model gets 3D surfaced; Expectation is just the selected edges are 3D surfaced.

Associated thread: https://forum.freecadweb.org/viewtopic.php?f=15&t=56981

# Additional information

The attached images demonstrate that the inclusion of face 7 to the 3d surface operation to complete the outside perimeter causes the entire model to be 3D surfaced.
Further commentary on the images may be found in the original thread https://forum.freecadweb.org/viewtopic.php?f=15&t=56981

# Steps to reproduce

Add all outside edges of a shape to by applied within a single 3D surface operation and apply.

# FreeCAD Info

```
(freecad-0.19-0.3.20201125gita50ae33557.fc33.2.x86_64)
OS: Fedora 33 (Workstation Edition) (GNOME/gnome)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.Unknown
Build type: Unknown
Python version: 3.9.0
Qt version: 5.15.1
Coin version: 4.0.0a
OCC version: 7.5.0
Locale: English/Australia (en_AU)
```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **Updated:** 4/13/2021

Russ4262, 2022-04-07:
0.20 pre-release analysis with Revision 28638:
 
The root cause of this bug is related to the current Millface bug not respecting the model: weakness in target shape formation.  While this clearly is a bug, the user is able to mill the selected faces in this case, just not in one smooth operation.  This fix will likely not be a mild fix, but I can only speculate.  Personally, I am not in a place to refactor anything to fix a bug like this.  My opinion is that priority is low with consideration to a pending 0.20 release.  However, I do feel the 3D Surface op is a strong component of the Path WB from a user perspective based on forum comments, and therefore might merit the attention to address this bug now.

Re: ballcutter segfault - I do not receive any error when adding Face 7 and computing the modified operation per bug report.

sliptonic, 2022-04-07:
Hm.  I'm getting the segfault even in trivial tests.  Just FYI, it's BULL cutter not BALL.  Any use of the bullcutter is segfaulting for me.

Russ4262, 2022-04-11:
Is the seg fault using the FC file provided in the forum link provided in this issue?  And, you say that 3D Surface is throwing a fault whenever the bull cutter is used, at any time?  Will you please post your FC info.  Perhaps this is OS specific, or python related?  Will you please post the seg fault error text.  Thanks.

My FC info for reference.
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.28671 +6 (Git)
Build type: Release
Branch: master_prs
Hash: 3ecfc7fc12fa2b311aa58e3c7f2e4ea49775d867
Python 3.8.6+, Qt 5.15.2, Coin 4.0.1, OCC 7.5.0
Locale: English/United States (en_US)
Installed mods: 
  * FC_SU
  * Z_MacroStartup
[/code]

luzpaz, 2022-07-22:
@sliptonic when you get a chance can you follow-up to russ's question?

maxwxyz, 2024-12-21:
@sliptonic @Russ4262 is this issue in CAM still relevant?

luzpaz, 2025-05-24:
Can someone reproduce this ?

FEA-eng, 2025-09-27:
@sliptonic Do you know if it's still relevant?

freckletonj, 2025-09-29:
It's still relevant in `v1.1 weekly`.