FreeCAD/FreeCAD #6200: .19 release crashes on creating certain surfaces

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4633

* **Reporter:** 56jimalexander
* **Date submitted:** 4/13/2021
* **FreeCAD version:** 0.19
* **Category:** Bug
* **Status:** new
* **Tags:** 

# Original report text

Original post with confirmation: https://forum.freecadweb.org/viewtopic.php?f=3&t=57538

This problem is well documented in the referenced post.
I am not able to upload the file (.2mb) here but it is available in the reference post.

# Steps to reproduce

See provided link

# FreeCAD Info

```

OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24212 (Git)
Build type: Release
Branch: master
Hash: 3d227cf7179c00e57f2733f947b4b1b0368d219a
Python version: 3.8.8
Qt version: 5.12.5
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)

```

# Other bug information

* **Priority:** high
* **Severity:** crash
* **Category:** Bug
* **Updated:** 4/13/2021

luzpaz, 2022-02-15:
@tomate44 do you mind weighing in on this?

VinceBa, 2022-02-24:
Still have the "access violation" pop up box.
...
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.27428 (Git)
Build type: Release
Branch: master
Hash: 27460358508a2057e0ec57a418641435f12628dd
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.3
Locale: French/France (fr_FR)

luzpaz, 2022-03-19:
@donovaly add to 0.20 milestone?

donovaly, 2022-03-19:
> @donovaly add to 0.20 milestone?

Yes, please tag all known crashes directly to this milestone.

donovaly, 2022-04-15:
Confirmed with latest master:

```
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.28697 (Git)
Build type: Release
Branch: master
Hash: 35bb6278ea4b93e8a9a67301042a75c065097373
Python 3.8.6+, Qt 5.15.2, Coin 4.0.1, OCC 7.5.3
Locale: English/United States (en_US) [ OS: German/Germany (de_DE) ]
```

AjinkyaDahale, 2022-05-15:
Judging from the reports, this could also have a `OS: Windows` tag.

donovaly, 2022-05-15:
Can anyone confirm the crash under Linux or MacOS?

donovaly, 2022-05-18:
This seems to be a bug in OpenCasCade: https://forum.freecadweb.org/viewtopic.php?p=595673#p595673

donovaly, 2022-05-26:
This will be fixed if OpenCascade is compiled without the CMake option "BUILD_RELEASE_DISABLE_EXCEPTIONS" (see https://wiki.freecadweb.org/Compile_on_Windows#Compiling_OpenCASCADE)

This was done for this LibPack: https://github.com/FreeCAD/FreeCAD-LibPack/releases/tag/2.4

So at least for the future Windows builds of FreeCAD 0.20, this crash will not occur.

luzpaz, 2022-05-26:
@donovaly what about other platforms/distros/binaries(appimage, snap etc...)

donovaly, 2022-05-26:
People have to build OCC with the CMake option BUILD_RELEASE_DISABLE_EXCEPTIONS turned OFF. Unfortunately OCC has it be default ON.

I will inform people in the forum release thread. I cannot do more.

luzpaz, 2022-05-26:
> People have to build OCC with the CMake option BUILD_RELEASE_DISABLE_EXCEPTIONS turned OFF. Unfortunately OCC has it be default ON.

#### Various FreeCAD packagers
@ppd  Snap
@greyltc  AUR
@looooo  conda/appimage
@hobbes1069  fedora
@PrzemoF  fedora copr
@waebbl  gentoo
@adrianschroeter  opensuse
@ipatch  homebrew
@hfiguiere  flatpak
@sgrogan  windows
@apeltauer  windows
@mandian   openmandriva

ppd, 2022-05-26:
@luzpaz We've had that since the very beginning for the edge builds: https://github.com/FreeCAD/freecad-deps-core20-snap/commit/1795c8791b058dbb3cb3d6353427e048fdffd353

adrianinsaval, 2022-05-27:
since it's OCC that needs to be compiled like that there's not much that can be done on the AUR side, just to keep you informed @luzpaz I'm comaintainer of the AUR package now, I've been thinking about creating an occ package with the blobfish patches though and if I do, I'll make it build with that option.

luzpaz, 2022-05-27:
Well played, sir! 👍

waebbl, 2022-05-27:
I can't open the attached file, so can currently not test with Gentoo.

```
Traceback (most recent call last):
  File "<string>", line 1, in <module>                                                                        
  File "/usr/lib64/freecad/Mod/Arch/importSH3D.py", line 47, in open                                          
    read(filename)                                                                                            
  File "/usr/lib64/freecad/Mod/Arch/importSH3D.py", line 79, in read                                          
    homexml = z.read("Home.xml")                                                                              
  File "/usr/lib/python3.9/zipfile.py", line 1472, in read                                                    
    with self.open(name, "r", pwd) as fp:                                                                     
  File "/usr/lib/python3.9/zipfile.py", line 1511, in open                                                    
    zinfo = self.getinfo(name)                                                                                
  File "/usr/lib/python3.9/zipfile.py", line 1438, in getinfo                                                 
    raise KeyError(                                                                                           
<class 'KeyError'>: "There is no item named 'Home.xml' in the archive"                                        
```

adrianinsaval, 2022-05-27:
are you using the latest version? I think the bug with Home.xml was fixed, did you deal with that @yorikvanhavre I can't remember the details.

Either way, what is needed for the bug here is to compile OCCT with `-D BUILD_RELEASE_DISABLE_EXCEPTIONS=OFF` can you check if that is the case for the gentoo script?

waebbl, 2022-05-27:
> are you using the latest version? I think the bug with Home.xml was fixed, did you deal with that @yorikvanhavre I can't remember the details.
> 

I first tried with the live ebuild (using latest git sources, built a week ago) and when this error showed up, I built 0.19.4 but with the same result.

> Either way, what is needed for the bug here is to compile OCCT with `-D BUILD_RELEASE_DISABLE_EXCEPTIONS=OFF` can you check if that is the case for the gentoo script?

This option isn't currently set for gentoo, we're using the default. I first wanted to check whether we're affected or not, before changing the build script. There was a post in this thread (or in the forum thread) which mentioned this might only affect windows systems, so I'd like to test first.

donovaly, 2022-05-28:
> which mentioned this might only affect windows systems

No, this affects all OSes. BUILD_RELEASE_DISABLE_EXCEPTIONS must be set OFF.

yorikvanhavre, 2022-05-28:
> are you using the latest version? I think the bug with Home.xml was fixed, did you deal with that @yorikvanhavre I can't remember the details.

this importSH3D error seems unrelated.. i'll have a look at it

waebbl, 2022-05-28:
> > which mentioned this might only affect windows systems
> 
> No, this affects all OSes. BUILD_RELEASE_DISABLE_EXCEPTIONS must be set OFF.

Thanks for this info. I"m going to update the opencascade build script then. Are all versions of OCCT affected? We currently have 7.5.2, 7.5.3, 7.6.0 and 7.6.1?

donovaly, 2022-05-28:
> Are all versions of OCCT affected?

Yes. This CMake option disables some of the exception handling of OCC but apparently we need it to be turned on.

adrianinsaval, 2022-05-28:
You probably meant off right?

waebbl, 2022-05-28:
The options needs to be turned off, which leads to exception handling being turned on ;)

adrianinsaval, 2022-05-28:
Oh that makes sense, silly me 😅

waebbl, 2022-05-28:
The Gentoo issue tracker for this: https://bugs.gentoo.org/847916

hfiguiere, 2022-05-28:
This was done on the flatpak, both 0.19 (stable) and 0.20beta1 (beta). occt 7.5.1 is used in both cases.

ipatch, 2022-08-19:
has anyone notified the opencascade devs about this "bug" ? i ask because i'm requesting the homebrew-core maintainers to enable this cmake flag for the homebrew-core version of opencascade?

https://github.com/Homebrew/homebrew-core/issues/108395

donovaly, 2022-08-22:
Not yet it seems. Can you please do this and report the issue to OpenCasCade?

luzpaz, 2022-09-26:
Has issue been reported upstream yet ?

donovaly, 2022-12-21:
> has anyone notified the opencascade devs about this "bug"

I don't see it as bug. You can of course compile OCC without some exception handling. The consequences are crashes. But why should the OCC guys change this? They might say you should know what you are doing.

-> closing this issue since the info is in our Wiki and OCC sets their defaults independently of FreeCAD's needs.

closed 2022-12-21 by donovaly