FreeCAD/FreeCAD #6201: Solver doesn't detect overconstraints

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4636

* **Reporter:** chrisb
* **Date submitted:** 4/17/2021
* **FreeCAD version:** 0.2
* **Category:** Bug
* **Status:** assigned
* **Tags:** 

# Original report text

See [forum post](https://forum.freecadweb.org/viewtopic.php?f=3&t=57661).

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **Updated:** 9/18/2021

# Discussion from Mantis ticket

### Comment by chrisb 2021-09-18 23:00
Here is another example of probably the same error: [Willem's example](https://forum.freecadweb.org/viewtopic.php?f=3&t=62241#p533501).

Zolko-123, 2022-10-25:
should be closed to clean up

luzpaz, 2022-10-30:
CC @PLChris

PLChris, 2022-12-09:
While I cannot reproduce with the file from the original forum topic, the issue with [Willem's file](https://forum.freecadweb.org/download/file.php?id=164639) still exists. The topic is linked above.

abdullahtahiriyo, 2022-12-10:
Thanks for checking Chris, and for mentioning there is a case where it can still be reproduced.

maxwxyz, 2024-03-09:
It's still existing. The equal constraint seems to go undetected.

FEA-eng, 2026-10-04:
[This](https://github.com/FreeCAD/FreeCAD/issues/6201#issuecomment-4116226345) is still relevant.

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 26.3.0dev.20260916 (Git shallow)
Build date: 2026/09/16 06:32:59
Build type: Release
Branch: (HEAD detached at a4ce44d33)
Hash: a4ce44d33b7e42e16cb8156058a1894237042ab4
Python 3.11.14, Qt 6.8.3, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.3.0, boost 1_84, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.0.0, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 617.14/NVIDIA Corporation/NVIDIA GeForce RTX 5080/PCIe/SSE2
```