FreeCAD/FreeCAD #6203: TechDraw not drawing all lines in view

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4639

* **Reporter:** run_the_race
* **Date submitted:** 4/20/2021
* **FreeCAD version:** 0.19
* **Category:** Bug
* **Status:** feedback
* **Tags:** #pending-forum, #ready-for-migration, OCC Bug

# Original report text

TechDraw is missing some lines in the isometric view

# Steps to reproduce

1. Open the attached file raspberry_pi_4b.FCStd
2. Open "Page" item in the tree browser
3. Look at the isometric view.
4. Notice some of the 40way header pins are missing vertical lines, as shown by the attached images.

# FreeCAD Info

```
OS: Windows 10 Version 2004
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24276 (Git)
Build type: Release
Branch: releases/FreeCAD-0-19
Hash: a88db11e0a908f6e38f92bfc5187b13ebe470438
Python version: 3.8.6+
Qt version: 5.15.1
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United Kingdom (en_GB)

```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **OS: Windows 10 2004**
* **Platform:** PC
* **Updated:** 1/24/2022

# Discussion from Mantis ticket

### Comment by Kunda1 2021-04-21 18:26
it's 'TechDraw' not 'tech draw'. Edited ticket.
I'll start a discussion for you on the forum.

---

### Comment by Kunda1 2021-04-21 18:27
Discussion at https://forum.freecadweb.org/viewtopic.php?f=35&t=57994

---

### Comment by wandererfan 2021-12-30 20:35
Some of the overlapping vertical lines for the pins are being treated as "hard hidden" instead of "hard visible".  Only part of the edge is hidden - does OCC not split the edge into visible and invisible portions before classifying??

---

### Comment by wandererfan 2022-01-02 17:18
is there a connection with 4564 short and/or overlapping lines not drawn??

---

### Comment by wandererfan 2022-01-06 18:26
using the perspective projector with a long focal length shows all the lines (also gives a better result for 4564).  

this has come up before, but I can't find an upstream ticket.

grubertechnic, 2023-05-15:
Can still confirm the issue. Even when i replace the rounding (they are only 3 digits precise) with precise expressions, the results is similar.

```
OS: Debian GNU/Linux 11 (bullseye) (X-Cinnamon/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33184 (Git)
Build type: Release
Branch: master
Hash: c33588be3e9e2bf555b133186a974d7dbd0b48f6
Python 3.9.2, Qt 5.15.2, Coin 4.0.0, Vtk 9.0.1, OCC 7.7.1
Locale: English/United States (en_US)
Installed mods: 
  * QuickMeasure 2022.10.28
  * Assembly4 0.12.6
  * CurvedShapes 1.0.4
  * Curves 0.6.8
```

[image]

I add the file here, that no one must search it : [raspberry_pi_4b.zip](https://github.com/FreeCAD/FreeCAD/files/11476560/raspberry_pi_4b.zip)

(small rant: now i know why i touched GH for so long, the writing, autoscrolling, auto view positioning and handling is a pita, what a mess)

WandererFan, 2023-06-01:
This works under the following conditions:
change Direction from (-0.5800000, -0.5800000, 0.5800000)
                            to     (-0.5800001, -0.5800000, 0.5800000)
                 
use coarse view with original params
use perspective view with original params

I think it has to do with the edges for the missing pins being partially overlapped by the pins in front of them.  

Since it works correctly with the polygon algo and perspective projector with the regular projector, I'm going to hypothesize that this is a problem with the occt hlr algo.

grubertechnic, 2023-06-04:
> I think it has to do with the edges for the missing pins being partially overlapped by the pins in front of them.

This is also what i think, a sort of (co)tangential OCCT issue.

stefano2734, 2026-09-12:
Increase your occt version to highest available. 7.5.0 is quite old here.
7.9.3 is last in 7er series with version 1.1.3 and 26.3 dev.
Also 8-series is supported by 26.3 dev.
8.0.1 is here actual. 8.02 and 8.1 will solve also FreeCAD problems.

WandererFan, 2026-09-12:
The missing lines are being classified incorrectly by HLRBRep_Algo/HLRBRep_HLRToShape as "hidden outline", but should be "visible outline".  This may be because the missing edges are partially hidden and should be split into a visible portion and a hidden portion?

Seems to be an upstream problem from here.

[image]

```
OS: Linux Mint 22.3 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 26.3.0dev.48516 +1 (Git)
Build date: 2026/09/09 14:22:03
Build type: Debug
Branch: LockedViewPositionOnLoad
Hash: 4f9963a87af18eabc919928e809c96f3396bc115
Python 3.12.3, Qt 6.11.0, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.1.0, boost 1_83, Eigen3 3.4.0, PySide 6.11.0
shiboken 6.11.0, SMESH 7.7.1.0, xerces-c 3.2.4, Clipper2 , IfcOpenShell 0.8.5, OCC 7.9.3
Locale: English/Canada (en_CA)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/
QPA/File dialog/Color dialog: xcb/via Qt/via Qt
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/129.405/1.09375
OpenGL version/vendor/renderer: 4.6 (Compatibility Profile) Mesa 25.2.8-0ubuntu0.24.04.2/Intel/Mesa Intel(R) UHD Graphics 620 (WHL GT2)
```

WandererFan, 2026-09-12:
See issue #32542 for another instance of misclassified edges.