FreeCAD/FreeCAD #6204: interactive controls in the viewport for all / most ( where applicable ) tools that have numeric values / inputs

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4640

* **Reporter:** nukeRomancer
* **Date submitted:** 4/21/2021
* **FreeCAD version:** 0.2
* **Category:** Feature
* **Status:** new
* **Tags:** 

# Original report text

interactive controls in the viewport for all / most ( where applicable ) tools that have numeric values / inputs

this would be a MASSIVE improvement to the general workflow

https://forum.freecadweb.org/viewtopic.php?p=496292#p496292

# Other bug information

* **Priority:** immediate
* **Severity:** feature
* **Category:** Feature
* **Updated:** 4/22/2021

# Discussion from Mantis ticket

### Comment by Pauvres_honteux 2021-04-22 14:51
[Some good examples of how user interacting stuff can be made, begins at 28 minutes](https://youtu.be/Ic0fxpypipg)

maxwxyz, 2024-03-09:
I've adjusted the original request for the topic mentioned at the FreeCAD Day 2024 Complaint Session:
**When you have done an operation, such as extrude, it’s impossible to visually edit that parameter via click and drag.**

pierreporte, 2024-04-29:
To really leverage this, feature preview is necessary: #11214.

Besides OVP-style controls, it would be useful to have an API to draw things like arrows, circles, lines, whatever to better show what is going to be created. Below is a screenshot from Catia showing how it could look. Elements can often be interacted with. For example, normal vectors displayed as an arrow can be inverted when clicked on.

[image]

ryanayoung, 2024-09-03:
Ran into another instance of this today. Was going to do feature request, but it falls under the umbrella of this request. Adjusting joint offsets in the assembly workbench. Currently there's no way to do it at all while creating the joint, it has to be done in the joint properties tab after the joint is created(at least for the slider joint).
I attached a screen recording of the process. and also attached a screenshot of the fusion equivalent.

https://github.com/user-attachments/assets/081c784b-be62-4ec4-af9d-dff758111be7

[image]