FreeCAD/FreeCAD #6205: Sketcher Clone - extend to allow substitution of dimensional constraints between different geometry items

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4641

* **Reporter:** jmaustpc
* **Date submitted:** 4/24/2021
* **FreeCAD version:** 0.2
* **Category:** Feature
* **Status:** assigned
* **Tags:** 

# Original report text

Sketcher Clone will currently substitute a dimensional constraint on a single geometry element with a "equals" constraint on the matching geometry element in the clone. 

The request is to extend this so that it will work when that dimensional constraint is between two different geometry elements. 

# Steps to reproduce

For example a Horizontal distance constraint applied to the top line of a rectangle will be an Equals constraint on the top line of the rectangle in the clone. However currently if the horizontal distance constraint for the same rectangle was applied between each of the top points of the two vertical side lines of the rectangle, then the same Horizontal constraint will be applied to the clone rectangle rather than an equals, therefore requiring editing every instance of the clone and original if an edit is required. 

Forum topic
https://forum.freecadweb.org/viewtopic.php?f=3&t=57915

An example file is attached to this ticket, it is the same file as one of the files in my post to that same topic in the link above.  

# FreeCAD Info

```
OS: Ubuntu 20.04.2 LTS (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.Stable PPA
Build type: Release
Branch: unknown
Hash: 9e3b630bbfb6807e93eeffb655e6c20f218644fc
Python version: 3.8.5
Qt version: 5.12.8
Coin version: 4.0.0
OCC version: 7.5.1
Locale: English/Australia (en_AU)

```

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 4/24/2021

adrianinsaval, 2023-03-28:
What could be done here is some sort of local carbon copy where cloned constraints will use an expression. The usefulness of making clones inside sketcher itself is questionable though, the general advice is to make patterns, copies etc. outside of sketcher.

abdullahtahiriyo, 2023-05-01:
There are several use cases, but yes, it could be extended to provide the option. As Adrian says, dimensional copies should be linked expressions (in one mode of operation). Then the user still has the option to remove the expression if he wants to scale the dimension. In another mode of operation, it can be the constraint substitution.

maxwxyz, 2024-02-21:
@PaddleStroke I think this is also implemented with #11267 and #11537