FreeCAD/FreeCAD #6206: Hotkeys working incorrectly for view mode in sketcher

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4644

* **Reporter:** run_the_race
* **Date submitted:** 4/25/2021
* **FreeCAD version:** 0.19
* **Category:** Bug
* **Status:** new
* **Tags:** 

# Original report text

In sketcher, display/render mode hotkeys dont work.

# Additional information

Forum discussion: [link]https://forum.freecadweb.org/viewtopic.php?f=3&t=58125[/link]

# Steps to reproduce

In Sketcher press `v, 3` to switch to shader mode wireframe, then `v, 1` to switch back to `normal`. It won't switch shader modes, instead it switches perspective.

# FreeCAD Info

```
OS: Windows 10 Version 2004
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24276 (Git)
Build type: Release
Branch: releases/FreeCAD-0-19
Hash: a88db11e0a908f6e38f92bfc5187b13ebe470438
Python version: 3.8.6+
Qt version: 5.15.1
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United Kingdom (en_GB)
```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **OS: Windows 10 2004**
* **Platform:** PC
* **Updated:** 12/29/2021

# Discussion from Mantis ticket

### Comment by chennes 2021-12-29 03:33
abdullah, where does this bug (and 0003796) stand now that you've reworked the Sketcher shortcuts?

luzpaz, 2022-06-06:
@abdullahtahiriyo if you don't mind taking a look at this?

abdullahtahiriyo, 2022-06-10:
@luzpaz 
As I see, it still happens exactly what the bug report indicates.
The issue is that "V" is the shortcut of "Vertical constraint". In the current configuration it is not possible to tackle it.

Now, there is a PR by RealThunder which I really really would like to see it merged, which would enable to differentiate between V and V,1. He uses a timer, so if the two are clicked within a time duration, then "V,1" is taken (and it should work just fine to do what the OP wants). If it times out then it is just "V" and then "1", which is what is happening now.

So when the feature freeze is removed, we could try to merge that one...

luzpaz, 2023-11-15:
@PaddleStroke here's another Sketcher shortcut related ticket that could get attention (since #11402 was tackled). Although it may not be as straight-forward.

abdullahtahiriyo, 2023-11-15:
@luzpaz 

This issue is fixed. My guess is that RT's PR was merged. Now "V" (Vertical constraint) lives happily with "V,1" and "V,3".

It may not be straight-forward to see it, but: Create one sketch on one plane (e.g., XY). Draw one line. Close. Create a second sketch on the same plane. Ensure the previous sketch is visible. You will see the line of the other sketch. Click "V", the "3" reasonably fast one after the other. The shade mode changes.

So the hotkey issue is solved.

closed 2023-11-15 by luzpaz