FreeCAD/FreeCAD #6208: [Arch] "Show Fill" in ArchView-TechDraw no longer works

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4646

* **Reporter:** bitacovir
* **Date submitted:** 4/27/2021
* **FreeCAD version:** 0.19
* **Category:** Bug
* **Status:** new
* **Tags:** 0.19, Arch

# Original report text

It seems Show Fill set to True no longer works for ArchWall in TechDraw with ArchView.
Forum discussion for more information: https://forum.freecadweb.org/viewtopic.php?f=23&t=57534
Bug in FC 0.19 and 0.20
Maybe assign this to Yorik.

# FreeCAD Info

```
OS: Windows 10 Version 1909
Word size of FreeCAD: 64-bit
Version: 0.20.24587 (Git)
Build type: Release
Branch: master
Hash: 1f62d4666102d8efd4d8f3ba58a1037456a3bcfb
Python version: 3.8.6+
Qt version: 5.15.1
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US)
```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **OS: windows 10**
* **Updated:** 4/27/2021

luzpaz, 2022-04-11:
Attached test file to ticket

luzpaz, 2022-09-13:
bumped forum thread

yorikvanhavre, 2023-05-16:
There were two problems here, that are solved by the above PR:
- The fill had a default white color so it was not noticeable. If the section plane's "UseMaterialColor" option is turned on, fills will now take the object color if no material is defined
- The fill didn't update when the object color changed. This is now solved too

PR #9585 (2023-05-23): Arch: Fixed SVG fill
fixes #6208 

Filling of cut objects was still happening, but it had an invisible, while default color and it was not updated when the object changed.

- [x]  Your Pull Request meets the requirements outlined in section 5 of [CONTRIBUTING.md](https://github.com/FreeCAD/FreeCAD/blob/master/CONTRIBUTING.md) for a Valid PR
commit 22226b31 (2023-05-23): Arch: Fixed SVG fill - fixes #6208
closed 2023-05-23 by yorikvanhavre