FreeCAD/FreeCAD #6211: Problem in using the sweep function

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4651

* **Reporter:** Daniele75
* **Date submitted:** 5/1/2021
* **FreeCAD version:** 0.19
* **Category:** Bug
* **Status:** feedback
* **Tags:** 

# Original report text

I found a bug when making springs that occurs in version 19, while up to 18 it worked smoothly.

!(https://ibb.co/1dN5CMq)

# Steps to reproduce

1° new file
2° goto "Part" ambient
3° creates parametric geometric primitives
4° Propeller
5° set dimension: pitch 0,75 ; height 3mm ; radius 1,3mm
6° Create Propeller ( Ok ) 
7° goto "Sketcher" ambient
8° Create new Sketcher
9° Select XZ floor
10° Create circle
11° Set constraints...  radius 0,2mm ; height from y axis 0mm ; distance from x axis 1,3mm ( same than radius of propeller )
12° close
13° goto "Part" ambient
14° select Sweep utilities
15° Select Sketch from list available profiles
16° bring to selected profiles
17° select all the propeller in the draw ( black to green color )
18° check create solid  + frenet
19° OK

the sweep obtained is corrupted.

# Other bug information

* **Priority:** high
* **Severity:** minor
* **Category:** Bug
* **OS: Windows 10 Home 20H2**
* **Updated:** 5/13/2021

# Discussion from Mantis ticket

### Comment by openBrain 2021-05-13 16:59
Daniele75 : please address your issue according [our reporting guidelines](https://forum.freecadweb.org/viewtopic.php?f=3&t=5236) by opening a new topic on [our forum](https://forum.freecadweb.org/index.php). Thx

plaes, 2023-11-26:
I ran into this issue when using 0.21.2 / git 33771 / b9bfa5c5507506e4515816414cd27f4851d00489 (Flatpak image).

It seems to be caused by certain multipliers for helix pitch (for example 3mm and 6mm):

Steps to reproduce:
1. Create helix: Pitch 6mm (or 3mm), Height 30mm, Radius: 8mm
2. Sketch a sweep pattern (circle on Object's XY plane, select helix as external geometry)
3. Sweep in Part workbench (Translation increment 2mm, rotation: 10 degrees)
4. Observe broken spring in 3D:
[image]
[image]

If I change Helix pitch to any other value (tried a lot different values from 2 .. 9mm), result looks ok:
[image]

Some other broken values: 3.01 and 6.02 (but not 6.01).

Part here compressed as ZIP:
[spring.zip](https://github.com/FreeCAD/FreeCAD/files/13466761/spring.zip)

luzpaz, 2023-11-26:
@plaes can you also test on the 0.22 development build ?

plaes, 2023-11-26:
> @plaes can you also test on the 0.22 development build ?

Yes, still present with latest one from https://github.com/FreeCAD/FreeCAD-Bundle:
>  FreeCAD 0.22.0, Libs: 0.22.0devR35209 (Git)

luzpaz, 2023-11-26:
@bgbsww want to take a swipe at this ?

luzpaz, 2023-11-27:
Thanks bgbsww for sleuthing this as far as you could. @FlachyJoe can you weigh-in ?

bgbsww, 2023-11-27:
~~Does that translation increment / rotation dialog actually do anything?  Or does it just come along for the ride as part of the ... --> ViewProviderExtrusion --> ViewProviderPart --> ViewProviderPartExt --> ViewProviderGeometryObject --> ViewProviderDragger inheritance path?~~  Oh, I see, that opens the transform widget, and those control what the transform widget does, but never affects the actual object directly.  Learned something today :)

bgbsww, 2023-11-30:
Yeah, I can't let go of this bone.  So, I think we're at this: 
https://github.com/FreeCAD/FreeCAD/blob/94ca51b7799b2ac60a2a56d7e6e753bb97a73671/src/Mod/Part/App/PartFeatures.cpp#L539-L541  and there are a bunch of setMode options: https://dev.opencascade.org/doc/refman/html/class_b_rep_offset_a_p_i___make_pipe_shell.html#a0e026a22a00587b3d36faffbcb68a2c9 
The simplest option would just be to force isFrenet, but I'm not sure about the cases where this would cause a problem.

I still don't really understand this, just adding data points.

bgbsww, 2023-11-30:
OK!  I'm not crazy.  @luzpaz assign me please, and we'll get this knocked out.

PR #11590 (2023-12-01): Default freenet to true for sweep to fix rendering bug
Fix #6211
closed 2023-12-01 by wwmayer