FreeCAD/FreeCAD #6212: Drilling Operation missing move to operation start depth.

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4652

* **Reporter:** dubstar_04
* **Date submitted:** 5/4/2021
* **FreeCAD version:** 0.19
* **Category:** Bug
* **Status:** new
* **Tags:** 

# Original report text

The drilling operation is missing a move to operation start depth

########## Example Output ##########
G0 Z35.000000
G90
***G1 X0.000000 Y0.000000 Z12.000000 <---- This line is missing***
G99
G81 R33.000000 X0.000000 Y0.000000 Z0.000000
G80
G0 Z33.000000
G81 R33.000000 X0.000000 Y0.000000 Z0.000000
G80
G0 Z33.000000
G0 Z35.000000
########## Example Output ##########

Forum Link: https://forum.freecadweb.org/viewtopic.php?f=15&t=58306

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **Updated:** 5/4/2021

Russ4262, 2022-03-13:
[[Path] Fixes for drilling operation](https://github.com/FreeCAD/FreeCAD/pull/4776#)
#4776
[sliptonic](https://github.com/sliptonic) merged 5 commits into [FreeCAD:master](https://github.com/FreeCAD/FreeCAD) from [dubstar-04:fixes/DrillingOp](https://github.com/dubstar-04/FreeCAD/tree/fixes/DrillingOp) on May 18, 2021

luzpaz, 2022-03-15:
Can this ticket be closed then?

Russ4262, 2022-03-15:
@luzpaz 

> Can this ticket be closed then?

My initial opinion is yes, but for a formal response I will defer to @J-Dunn and @sliptonic for that answer.

closed 2022-03-15 by sliptonic

J-Dunn, 2022-03-15:
Yep, I have not seem the same anomalies since Russ's fix.  Thanks for Russ for fixing this small but rather important bug.

Russ4262, 2022-03-15:
@J-Dunn I don't think I had anything to do with it. ;)

J-Dunn, 2022-03-15:
Ah , apologies to dub'  , I did not read your comment properly.

Russ4262, 2022-03-15:
And my apology to @dubstar-04 as well. I was confused on users. Thanks Dubstar-04 for the fix.