FreeCAD/FreeCAD #6214: Problem to rotate 3D-view with 3D-mouse

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4654

* **Reporter:** SeppoTu
* **Date submitted:** 5/4/2021
* **FreeCAD version:** 0.19
* **Category:** Bug
* **Status:** new
* **Tags:** 3dconnexion, mouse, rotate, Win

# Original report text

Center of rotation is not updated in perspective view mode, during transfer with 3D-mouse.
In perspective view mode the normal mouse operations sets rotating center in correct postion.
Using 3D-mouse the rotation center is not updated. This makes it very difficult to use 3D-mouse in prespective view mode.
Tested with 3DConnexion's SpacePilot Pro.

# Steps to reproduce

1. Open FreeCAD.
2. Create new part (cube).
4. Change view mode to perspective view.
5. Rotate cube with 3D-mouse. Rotation center is in cube's center.
6. Move cube away from it's original position by using 3D-mouse.
7. Rotate cube again with 3D-mouse. Rotation center has not transferred with the cube.
   Instead rotation center is still at the old location.
8. Rotate cube with normal mouse. Cube jumps to to it's original location (rotation center) before it starts rotate.

# FreeCAD Info

```
OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24276 (Git)
Build type: Release
Branch: releases/FreeCAD-0-19
Hash: a88db11e0a908f6e38f92bfc5187b13ebe470438
Python version: 3.8.6+
Qt version: 5.15.1
Coin version: 4.0.1
OCC version: 7.5.0
Locale: Finnish/Finland (fi_FI)

```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **OS: Windows 10 Pro 20H2**
* **Platform:** Table PC
* **Updated:** 5/5/2021

# Discussion from Mantis ticket

### Comment by SeppoTu 2021-05-05 07:45
Correction to reproduction:
In reproduce step 6, I ment to say that: zoom the view with 3D-mouse, not that cube it self has to move.

adrianinsaval, 2023-03-28:
is this still relevant? anybody else with a 3d mouse can check this?

luzpaz, 2023-03-28:
@chennes, you do right ?

chennes, 2023-03-28:
Yeah - I don't think I am reproducing this exact behavior, but I definitely get some rotation glitches in perspective view, with sudden jumps between angles. Maybe I'm just garbage at using the mouse (I am still new at it), but it doesn't seem right to me.

FEA-eng, 2024-06-29:
This seems to be fixed by #12929 but it would be nice if someone could confirm.

chennes, 2024-06-29:
I just checked, and there is still some kind of problem switching from SpaceMouse to normal mouse.  If I follow the above instructions through step 6, it appears that the rotation center indicator DOES move with the cube. But when I switch to the normal mouse, the cube still jumps in position (a zoom, perhaps?) as soon as I begin a rotation.

luzpaz, 2024-10-27:
Can someone re-test this ?

FEA-eng, 2024-10-27:
> Can someone re-test this ?

What Chris mentioned (jump when switching from 3D mouse to regular mouse) is still relevant.

[image]

luzpaz, 2025-01-26:
@oficinerobotica can you test to see if this is still relevant on your 3dconnexion mouse ?

FEA-eng, 2025-01-26:
I don't get the jump when switching to a normal mouse now. But the part is shaking strangely (it doesn't happen in 1.0):

[image]

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39980 (Git) Conda
Build type: Release
Branch: main
Hash: 836ad91f3ddd3ec63b84b8e3f87a767d5465261d
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

oficinerobotica, 2025-01-27:
I can't seem to reproduce the jumping nor the shaking but I tested on Linux atm, and the space mouse driver is limited respect to the windows version.  @FEA-eng, On the spacemouse side,  do you  have "lock horizon" enabled by chance? there are so many variables envolved that could be tested to try to narrow down the problem

https://github.com/user-attachments/assets/7e846e5a-3008-491a-93aa-a45495d635fc

FEA-eng, 2026-09-23:
Needs feedback from @chennes.