FreeCAD/FreeCAD #6215: [Arch] Double click on door/window opens BIM wokbench

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4655

* **Reporter:** bitacovir
* **Date submitted:** 5/4/2021
* **FreeCAD version:** 0.19
* **Category:** Bug
* **Status:** new
* **Tags:** 0.19, Arch

# Original report text

When I double click on Arch door/window, in the combo view panel, FreeCAD switches to BIM wokbench. If you are in any workbench, it'll change to BIM with the double click (tested on 0.19.2 and 0.20)

Forum discussion:
https://forum.freecadweb.org/viewtopic.php?f=23&t=58362

# Steps to reproduce

Open Arch workbench
Create a Arch wall
Insert an Arch Window in the wall
Double click on the window object in the combo view panel
The workbench changes to BIM workbench (that I have installed).

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **OS: Windows 10**
* **Updated:** 5/13/2021

luzpaz, 2023-02-27:
@yorikvanhavre should be transferred to BIM workbench issue queue

yorikvanhavre, 2023-02-28:
This is weird... Having a look

yorikvanhavre, 2023-02-28:
I can't reproduce the issue... I think it must have been a temporary thing and is now solved.
@bitacovir can you test again?

luzpaz, 2023-04-15:
Closing. If issue returns when you test @bitacovir, please re-open.

closed 2023-04-15 by luzpaz