FreeCAD/FreeCAD #6216: Vertices turn black with box selection when no constraints are visible in first new sketch

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4657

* **Reporter:** mcv100
* **Date submitted:** 5/4/2021
* **FreeCAD version:** 0.19
* **Category:** Bug
* **Status:** new
* **Tags:** 

# Original report text

I'm seeing a strange issue in the Sketcher, where all vertices (including the origin) turn black after a box selection, unless there are constraint symbols visible. I can reproduce this issue consistently and very easily, even on a fresh 0.19.2 install with all preferences reset. 

Just make a new file, go into part design, new body, new sketch and make a box selection. I see the origin turn black instantly (even if it was selected), unless there were constraints visible it seems.

I cannot reproduce this issue on a new body, new sketch if the file already contains other bodies. Only in a fresh file does this happen for me.

Forum link:
https://forum.freecadweb.org/viewtopic.php?f=3&t=58314&p=501224

[DEV EDIT] more reports on the forum :
https://forum.freecadweb.org/viewtopic.php?f=3&t=57744
https://forum.freecadweb.org/viewtopic.php?f=12&t=55260

# Steps to reproduce

1. Make new file
2. Go into part design
3. Make new body
4. Start a new sketch
5. Make a box selection
6. See origin turn black

# FreeCAD Info

```
OS: Windows 10 Version 1909
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24291 (Git)
Build type: Release
Branch: releases/FreeCAD-0-19
Hash: 7b5e18a0759de778b74d3a5c17eba9cb815035ac
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/Netherlands (en_NL)
```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **OS: Windows 10 1909**
* **Platform:** Windows
* **Updated:** 5/6/2021

luzpaz, 2022-05-08:
Per @PLChris in the [forum thread](https://forum.freecadweb.org/viewtopic.php?p=496384#p496384)

> I remember Werner saying that something is running out of resources. Reopening sketcher or even restarting FreeCAD usually helps.

Aside: I've updated the GH ticket to show the screenshots.

luzpaz, 2024-09-08:
This looks similar to #11444 #11541

luzpaz, 2024-12-23:
bumping for discussion

closed 2025-08-20 by github-actions[bot]