FreeCAD/FreeCAD #6218: Select Color -> Pick Screen Color cannot select a colour outside the dialog window

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4659

* **Reporter:** samantha-uk
* **Date submitted:** 5/18/2021
* **FreeCAD version:** 0.19
* **Category:** Bug
* **Status:** new
* **Tags:** UI

# Original report text

When using Select Color -> Pick Screen Color, the selection pointer is restricted to sampling inside the dialog box.
!(https://forum.freecadweb.org/download/file.php?id=153186)

This contrasts with the Spreadsheet color picker where the eyedropper tool can sample anywhere on the screen.
!(https://forum.freecadweb.org/download/file.php?id=153185)

# Additional information

Contrast with the similar activity from spreadsheet workbench, which seems to use a different approach.

# Steps to reproduce

Fire up the Select Color dialog from Preferences or Property View/view and select the Pick Screen Color feature.
The cursor will not move outside the dialog boundary.

# Other bug information

* **Priority:** normal
* **Severity:** trivial
* **Category:** Bug
* **OS: OSX Catalina 10.15.7**
* **Platform:** Macbook Intel
* **Updated:** 5/18/2021

luzpaz, 2022-05-03:
@PLChris would you mind confirming this on macos?

luzpaz, 2022-05-03:
I can'r reproduce this on 
```
OS: Ubuntu Core 20 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.28786 (Git)
Build type: Release
Branch: master
Hash: ccc8ac65c5a04f7f6690c29dc08225216f201e85
Python 3.8.10, Qt 5.15.3, Coin 4.0.0, OCC 7.6.1
Locale: English/United States (en_US)
```

PAS-DRAFTER, 2022-10-24:
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.30716 (Git)
Build type: Release
Branch: master
Hash: cbd19631428f1d9eb4faae30220551608a648589
Python 3.8.13, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.12.3
  * Curves 0.5.6
  * ExplodedAssembly
  * fasteners 0.4.2
  * Manipulator 1.4.9
  * MOOC 2022.4.21
  * OpticsWorkbench 1.0.8
  * sheetmetal 0.2.56
  * ToolViewReverse_ToolBar
[/code]

I was having problems with the select color working on the v20 release as well.  I just downloaded the latest weekly build and thought it was working.  However, I found that it worked on my secondary screen, but not my primary one.  On my secondary screen, as I moved my select cursor across the screen, the color would change, but it wouldn't change to the one that was under the cursor.  As I moved the cross hairs up and down on the screen, I would see the colors change but not at the right spot.  My displays have two different screen resolutions, maybe that has something to do with it?

PAS-DRAFTER, 2022-10-31:
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.30716 (Git)
Build type: Release
Branch: master
Hash: cbd19631428f1d9eb4faae30220551608a648589
Python 3.8.13, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.12.3
  * Curves 0.5.6
  * ExplodedAssembly
  * fasteners 0.4.2
  * Manipulator 1.4.9
  * MOOC 2022.4.21
  * OpticsWorkbench 1.0.8
  * sheetmetal 0.2.56
  * ToolViewReverse_ToolBar
[/code]
Just checked again this morning on my other work station.  The color picker would work on my lap top screen properly, but not on my 2nd display.  In both cases where it didn't work properly, it was on my right side screen.  At work that was my laptop screen and at home that is my 2nd display.  So I don't think it is related to primary an secondary displays.

PLChris, 2022-12-09:
I can confirm the issue.
There is a workaround:
- Place the mouse over the place from which you want to pick the color
- invoke the picking with the keyboard
- terminate the picking with the keyboard's ENTER

luzpaz, 2024-08-04:
@JohnOCFII can you reproduce this on latest dev ?

JohnOCFII, 2024-08-05:
> @JohnOCFII can you reproduce this on latest dev ?

This isn't a feature I've used before, but reading the issue, if the correct behavior is that the color picker ought to be able to select colors from anywhere in the FreeCAD window (and not just restricted to the color dialog) then I'd say I could not reproduce the issue, and the system is now working as expected.

```
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38385 (Git)
Build type: Release
Branch: main
Hash: 58241e027e7187a1ecaeeb75668ef9a497437c68
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.7.30
```