FreeCAD/FreeCAD #6219: FreeCAD crashes on spreadsheet update Due to Fillet

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4661

* **Reporter:** BruceL
* **Date submitted:** 5/19/2021
* **FreeCAD version:** 0.19
* **Category:** Bug
* **Status:** assigned
* **Tags:** #ready-for-migration, fillet, OCC 7.6, OCC Bug, upstream

# Original report text

[Forum Post](https://forum.freecadweb.org/viewtopic.php?f=3&t=58687&p=504394#p504394)

FreeCAD crashes on spreadsheet update Due to Fillet

# Additional information

Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x3bd60) [0x7f6b54484d60]
#1  0x7f6b2b362b98 in Geom2d_Curve::Value(double) const from /tmp/.mount_FreeCA0weYfg/usr/lib/./libTKG2d.so.7+0x18
#2  0x7f6b2d505b46 in ChFi3d_Builder::PerformMoreThreeCorner(int, int) from /tmp/.mount_FreeCA0weYfg/usr/lib/./libTKFillet.so.7+0xa936
#3  0x7f6b2d4e701e in ChFi3d_Builder::PerformIntersectionAtEnd(int) from /tmp/.mount_FreeCA0weYfg/usr/lib/./libTKFillet.so.7+0x7cfe
#4  0x7f6b2d4eae1a in ChFi3d_Builder::PerformOneCorner(int, bool) from /tmp/.mount_FreeCA0weYfg/usr/lib/./libTKFillet.so.7+0x10ba
#5  0x7f6b2d4874fe in ChFi3d_Builder::PerformFilletOnVertex(int) from /tmp/.mount_FreeCA0weYfg/usr/lib/./libTKFillet.so.7+0x51e
#6  0x7f6b2d4884fc in ChFi3d_Builder::Compute() from /tmp/.mount_FreeCA0weYfg/usr/lib/./libTKFillet.so.7+0xe5c
#7  0x7f6b2d64d504 in BRepFilletAPI_MakeFillet::Build() from /tmp/.mount_FreeCA0weYfg/usr/lib/./libTKFillet.so.7+0x34
#8  0x7f6af984bbb1 in PartDesign::Fillet::execute() from /tmp/.mount_FreeCA0weYfg/usr/lib/_PartDesign.so+0x4b1
#9  0x7f6b56269ec2 in App::DocumentObject::recompute() from /tmp/.mount_FreeCA0weYfg/usr/bin/../lib/libFreeCADApp.so+0x122
#10  0x7f6b2e48962a in Part::Feature::recompute() from /tmp/.mount_FreeCA0weYfg/usr/lib/Part.so+0xa
#11  0x7f6b561fa81d in App::Document::_recomputeFeature(App::DocumentObject*) from /tmp/.mount_FreeCA0weYfg/usr/bin/../lib/libFreeCADApp.so+0x2fd
#12  0x7f6b56255dd6 in App::Document::recompute(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool, bool*, int) from /tmp/.mount_FreeCA0weYfg/usr/bin/../lib/libFreeCADApp.so+0x5c6
#13  0x7f6b562b557f in App::DocumentPy::recompute(_object*) from /tmp/.mount_FreeCA0weYfg/usr/bin/../lib/libFreeCADApp.so+0x2df
#14  0x7f6b562b5749 in App::DocumentPy::staticCallback_recompute(_object*, _object*) from /tmp/.mount_FreeCA0weYfg/usr/bin/../lib/libFreeCADApp.so+0x29
#15  /tmp/.mount_FreeCA0weYfg/usr/bin/../lib/libpython3.8.so.1.0(PyCFunction_Call+0xf5) [0x7f6b566b7725]
#16  /tmp/.mount_FreeCA0weYfg/usr/bin/../lib/libpython3.8.so.1.0(_PyObject_MakeTpCall+0x230) [0x7f6b5665f550]
#17  /tmp/.mount_FreeCA0weYfg/usr/bin/../lib/libpython3.8.so.1.0(_PyEval_EvalFrameDefault+0x4ddd) [0x7f6b566c928d]
#18  /tmp/.mount_FreeCA0weYfg/usr/bin/../lib/libpython3.8.so.1.0(_PyEval_EvalCodeWithName+0x2ea) [0x7f6b56660d9a]
#19  /tmp/.mount_FreeCA0weYfg/usr/bin/../lib/libpython3.8.so.1.0(PyEval_EvalCodeEx+0x3a) [0x7f6b566ee40a]
#20  /tmp/.mount_FreeCA0weYfg/usr/bin/../lib/libpython3.8.so.1.0(PyEval_EvalCode+0x1c) [0x7f6b566ee42c]
#21  /tmp/.mount_FreeCA0weYfg/usr/bin/../lib/libpython3.8.so.1.0(+0x1e05aa) [0x7f6b567135aa]
#22  /tmp/.mount_FreeCA0weYfg/usr/bin/../lib/libpython3.8.so.1.0(+0x224225) [0x7f6b56757225]
#23  /tmp/.mount_FreeCA0weYfg/usr/bin/../lib/libpython3.8.so.1.0(PyRun_StringFlags+0x80) [0x7f6b567572c0]
#24  0x7f6b55f542cc in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /tmp/.mount_FreeCA0weYfg/usr/bin/../lib/libFreeCADBase.so+0x6c
#25  0x7f6b56d6e091 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /tmp/.mount_FreeCA0weYfg/usr/bin/../lib/libFreeCADGui.so+0x91
#26  0x7f6b56d6e231 in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /tmp/.mount_FreeCA0weYfg/usr/bin/../lib/libFreeCADGui.so+0x101
#27  0x7f6af8961365 in SpreadsheetGui::SheetModel::setData(QModelIndex const&, QVariant const&, int) from /tmp/.mount_FreeCA0weYfg/usr/lib/SpreadsheetGui.so+0x105
#28  0x7f6af8955e30 in SpreadsheetGui::SheetView::editingFinished() from /tmp/.mount_FreeCA0weYfg/usr/lib/SpreadsheetGui.so+0x140
#29  0x7f6b5497ca18 in QMetaObject::activate(QObject*, int, int, void**) from /tmp/.mount_FreeCA0weYfg/usr/bin/../lib/libQt5Core.so.5+0x6c0
#30  0x7f6af89478f7 in SpreadsheetGui::LineEdit::event(QEvent*) from /tmp/.mount_FreeCA0weYfg/usr/lib/SpreadsheetGui.so+0x127
#31  0x7f6b5518c7f2 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCA0weYfg/usr/bin/../lib/libQt5Widgets.so.5+0xdc
#32  0x7f6b55190cbb in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCA0weYfg/usr/bin/../lib/libQt5Widgets.so.5+0x467
#33  0x7f6b56d3cdf9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCA0weYfg/usr/bin/../lib/libFreeCADGui.so+0x89
#34  0x7f6b54966606 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCA0weYfg/usr/bin/../lib/libQt5Core.so.5+0x86
#35  /tmp/.mount_FreeCA0weYfg/usr/bin/../lib/libQt5Widgets.so.5(+0x1aff6c) [0x7f6b551c7f6c]
#36  0x7f6b5518c7f2 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCA0weYfg/usr/bin/../lib/libQt5Widgets.so.5+0xdc
#37  0x7f6b5519231d in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCA0weYfg/usr/bin/../lib/libQt5Widgets.so.5+0x1ac9
#38  0x7f6b56d3cdf9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCA0weYfg/usr/bin/../lib/libFreeCADGui.so+0x89
#39  0x7f6b54966606 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCA0weYfg/usr/bin/../lib/libQt5Core.so.5+0x86
#40  0x7f6b54d08eaa in QGuiApplicationPrivate::processKeyEvent(QWindowSystemInterfacePrivate::KeyEvent*) from /tmp/.mount_FreeCA0weYfg/usr/bin/../lib/libQt5Gui.so.5+0xca
#41  0x7f6b54d0b939 in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) from /tmp/.mount_FreeCA0weYfg/usr/bin/../lib/libQt5Gui.so.5+0x79
#42  0x7f6b54cf8c65 in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCA0weYfg/usr/bin/../lib/libQt5Gui.so.5+0x63
#43  /tmp/.mount_FreeCA0weYfg/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5(+0x5b010) [0x7f6b501b2010]
#44  /tmp/.mount_FreeCA0weYfg/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_dispatch+0x27d) [0x7f6b518e123d]
#45  /tmp/.mount_FreeCA0weYfg/usr/bin/../lib/./libglib-2.0.so.0(+0x554d1) [0x7f6b518e14d1]
#46  /tmp/.mount_FreeCA0weYfg/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x31) [0x7f6b518e1571]
#47  0x7f6b549950e8 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCA0weYfg/usr/bin/../lib/libQt5Core.so.5+0x5e
#48  0x7f6b54963019 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCA0weYfg/usr/bin/../lib/libQt5Core.so.5+0x187
#49  0x7f6b54966e8f in QCoreApplication::exec() from /tmp/.mount_FreeCA0weYfg/usr/bin/../lib/libQt5Core.so.5+0xfd
#50  0x7f6b56cc6926 in Gui::Application::runApplication() from /tmp/.mount_FreeCA0weYfg/usr/bin/../lib/libFreeCADGui.so+0x15f6
#51  /tmp/.mount_FreeCA0weYfg/usr/bin/freecad(+0x44bf) [0x55a8337464bf]
#52  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xea) [0x7f6b5446fd0a]
#53  /tmp/.mount_FreeCA0weYfg/usr/bin/freecad(+0x47d9) [0x55a8337467d9]

# Steps to reproduce

On the attached file, open the spreadsheet.

Change the formula in cell B6 from

`=board_depth + 2 * (wall_width + part_clearance)`

to
```
=board_depth + 2 * (thin_wall_width + part_clearance)```

FreeCAD will crash with the output attached below.

I'm attempting to change the depth of the "case" pad by changing the formula as listed above.

# Other bug information

* **Priority:** low
* **Severity:** crash
* **Category:** Bug
* **Updated:** 1/24/2022

# Discussion from Mantis ticket

### Comment by BruceL 2021-05-19 01:22
Trying again to upload the file, I'm informed that there's no space on the webserver.

---

### Comment by BruceL 2021-05-19 01:23
I'm not able to upload the file to the bug tracker. It's available on the forum [here](https://forum.freecadweb.org/download/file.php?id=153239)

---

### Comment by chennes 2021-08-17 16:57
Can you still reproduce this on the latest HEAD? I am not getting a crash right now.

---

### Comment by BruceL 2021-08-19 21:22
I can report that the crash still occurs on the latest weekly Appimage (25429).

I'll see what's involved in running the latest HEAD version.

---

### Comment by chennes 2021-08-20 15:13
If the AppImage is compiled against the official OpenCASCADE libraries, rather than the "blobfish" patched version, then we'll just have to wait for OCCT 7.6 to be released to fix this.

---

### Comment by BruceL 2021-08-20 21:03
I suspect that's the case, the current Appimage shows it using the OCC 7.5.2 libraries.

adrianinsaval, 2022-02-10:
I don't get a crash in any of these versions:
```
OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24365 (Git)
Build type: Release
Branch: releases/FreeCAD-0-19
Hash: c5eca93a2070fe67957350ef8371e53027364c69
Python version: 3.8.6+
Qt version: 5.15.1
Coin version: 4.0.1
OCC version: 7.5.2
Locale: Spanish/Paraguay (es_PY)
```
```
OS: Windows 10 (10.0)
Word size of FreeCAD: 64-bit
Version: 0.20.26761 (Git)
Build type: Release
Python version: 3.8.12
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.3
Locale: Spanish/Paraguay (es_PY)
```
should we consider this fixed and close the issue?

chennes, 2022-02-10:
Yes, especially now that OCCT 7.6 is available, which has the official upstream fix. Anyone still experiencing this issue needs to change to a version compiled against either the latest OCCT, or against the "blobfish" patchset.