FreeCAD/FreeCAD #6220: Undo does not reset the document saved state

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4666

* **Reporter:** tarun.johar
* **Date submitted:** 5/24/2021
* **FreeCAD version:** 0.19
* **Category:** Bug
* **Status:** new
* **Tags:** 

# Original report text

Undo does not reset the document saved state. The issue is reproducible with version 0.19 and 0.19.2

The cylinder disappears but there is still a '*' in the document tab.

We are developing a workbench that creates temporary objects for preview purposes. When an operation is canceled, the transaction is properly aborted but the document saved state is not reverted. Applications such as Emacs and LibreOffice keep track of the saved state as part of the undo stack, so this is a reasonable expectation for the user.

# Additional information

The topic is already discussed at following Help on using FreeCAD forum link : viewtopic.php?f=3&t=58752&e=1&view=unread#unread

# Steps to reproduce

Steps to reproduce:

1. Create a new document
2. Create a Part Design box
3. Save the document
4. There is no '*' next to the document name
5. Create a Part Design cylinder
6. There is a '*' next to the document name
7. Select Edit | Undo

# FreeCAD Info

```
OS: Ubuntu 20.04.2 LTS (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24291 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.19.2)
Hash: 7b5e18a0759de778b74d3a5c17eba9cb815035ac
Python version: 3.8.8
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)

```

# Other bug information

* **Priority:** normal
* **Severity:** major
* **Category:** Bug
* **Updated:** 12/9/2021

# Discussion from Mantis ticket

### Comment by tarun.johar 2021-05-28 18:50
Hi team,

Can this be looked into?

---

### Comment by wmayer 2021-12-09 11:50
https://forum.freecadweb.org/viewtopic.php?f=3&t=58752&p=504797#p504797

luzpaz, 2022-04-03:
@donovaly would you consider this bug important for v0.20 ?

donovaly, 2022-04-04:
Since there is no crash nor regression and also no dataloss, I don't see it as mustfix.

luzpaz, 2025-01-28:
This bug introduces an unnecessary 'dirtying' of the history of a document that one might have opened and moved around it.  It seems like important etiquette to tighten up this functionality because it could lead to users becoming paranoid about what they might of changed in a document without fully knowing for sure because they can't revert the document to a clean untouched state via Undo.