FreeCAD/FreeCAD #6222: Datum (mm) for the constraint with index 14) is invalid

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4670

* **Reporter:** TychoPrince
* **Date submitted:** 5/29/2021
* **FreeCAD version:** 0.19
* **Category:** Bug
* **Status:** feedback
* **Tags:** #pending-forum

# Original report text

This same error code appears for horizontal distance editing also but refers to index 11

# Steps to reproduce

Sketch a rectangle. Sketch a line from the corner of a rectangle parallel to a vertical edge and attempt to edit the vertical distance.

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **Updated:** 11/19/2021

# Discussion from Mantis ticket

### Comment by chennes 2021-08-11 21:52
I cannot reproduce this error: is there as associated forums discussion?

ryan7653, 2023-02-16:
I'm having this bug right now

adrianinsaval, 2023-02-16:
you need to provide a lot more details than that if you want any serious response, full version info, reproduction steps, good description of the problem...