FreeCAD/FreeCAD #6223: Datum 10,00 m for the constraint with index 6 is invalid

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4672

* **Reporter:** wst
* **Date submitted:** 5/31/2021
* **FreeCAD version:** 0.19
* **Category:** Bug
* **Status:** feedback
* **Tags:** 

# Original report text

1.  create a new file > new part > open sketcher.
2.  draw a vertical line (vertical constraint)
3.  add a vertical length constraint of 4200 mm
4.  draw a new horizontal line from the bottom point to the right (horizontal constraint)
5.  add a horizontal length constraint of 10000 mm (10m)

You will receive: "Datum 10,00 m for the constraint with index 6 is invalid.

```
12:28:18  Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ValueError'>: Datum 10,00 m for the constraint with index 6 is invalid
12:28:18  App.getDocument('Naamloos').getObject('Sketch').setDatum(6,App.Units.Quantity('10000.000000 mm'))
12:28:30  Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ValueError'>: Datum 10,00 m for the constraint with index 6 is invalid
12:28:30  App.getDocument('Naamloos').getObject('Sketch').setDatum(6,App.Units.Quantity('10000.000000 mm'))
```

# Additional information

Windows 10 latest update

# Steps to reproduce

1.  create a new file > new part > open sketcher.
2.  draw a vertical line (vertical constraint)
3.  add a vertical length constraint of 4200 mm
4.  draw a new horizontal line from the bottom point to the right (horizontal constraint)
5.  add a horizontal length constraint of 10000 mm (10m)

# FreeCAD Info

```
OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24276 (Git)
Build type: Release
Branch: releases/FreeCAD-0-19
Hash: a88db11e0a908f6e38f92bfc5187b13ebe470438
Python version: 3.8.6+
Qt version: 5.15.1
Coin version: 4.0.1
OCC version: 7.5.0
Locale: Dutch/Belgium (nl_BE)

```

# Other bug information

* **Priority:** high
* **Severity:** major
* **Category:** Bug
* **Updated:** 6/2/2021

# Discussion from Mantis ticket

### Comment by Kunda1 2021-06-02 18:56
wst can you reproduce on v0.20 dev? https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

adrianinsaval, 2022-02-10:
I can't reproduce, no forum post and no further comment from OP -> should be closed IMO

closed 2022-02-10 by chennes

regtm, 2022-03-01:
I experience the same error when redundant constraints are present. Resolving the redundancies first solves the issue.

TNTRedstone, 2023-03-26:
I had the same issue but, can't find any redundancies in my sketch.

My sketch:
[image]

adrianinsaval, 2023-03-26:
For help request to to the help forum and post your file. forum.freecad.org