FreeCAD/FreeCAD #6226: Programmatically building sketch geometry and constraints displays error, but creates valid geometry and constraints

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4675

* **Reporter:** drmacro
* **Date submitted:** 6/4/2021
* **FreeCAD version:** 0.2
* **Category:** Bug
* **Status:** confirmed
* **Tags:** #pending-forum, python, sketcher

# Original report text

Adding this as requested by abdullah

When programmatically building geometry the following error is displayed, but, only with certain sets of valid input data.

```Updating geometry: Error build geometry(6): gp_Circ::SetRadius() - radius should be positive number
Invalid solution from DogLeg solver.```

It appears to only happen with certain sets of data in the Python scripts available here:
(https://github.com/macdroid53/FreeCADScripts)
(put the files in the macro directory and execute the .py file with no _ui in the name. 

[Forum discussion.](https://forum.freecadweb.org/viewtopic.php?f=22&t=58566)

# Steps to reproduce

Using certain set of data input to the linked Python scripts produces the error shown.

Sometimes all values appear to work. Sometimes not. And at other times the same values produce no error.

I have been unable to discern a pattern.

Sometimes the error mentions multiple solvers.

# FreeCAD Info

```
OS: Ubuntu 20.04.2 LTS (XFCE/ubuntustudio)
Word size of FreeCAD: 64-bit
Version: 0.20.25024 (Git)
Build type: Release
Branch: master
Hash: 1fb45a0c5ee88ddd31c1e17b2b3efc83fc9106f6
Python version: 3.8.5
Qt version: 5.12.8
Coin version: 4.0.0
OCC version: 7.5.1
Locale: English/United States (en_US)

```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **Updated:** 2/3/2022

luzpaz, 2022-04-14:
CC @mnesarco are you encountering this at all in your scripting API explorations?

mnesarco, 2022-04-14:
@luzpaz nop. At least not yet.

luzpaz, 2022-04-14:
CC @macdroid53 since you're mentioned in the original ticket

luzpaz, 2022-05-03:
CC @onekk (participating in the thread)

onekk, 2022-05-03:
Thanks for the CC, but Sketcher is not my most known WB of FC.
My comments on thread was more general, sadly after some time (almost 1 year from the last post) my knowledge has not improved in such field.

Regards

Carlo D.

closed 2024-12-29 by github-actions[bot]