FreeCAD/FreeCAD #6227: Not have Std_DuplicateSelection/Std_Copy automatically check dependencies

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4676

* **Reporter:** TedM
* **Date submitted:** 6/5/2021
* **FreeCAD version:** 0.19
* **Category:** Feature
* **Status:** new
* **Tags:** 

# Original report text

Currently, when doing Std_DuplicateSelection (Edit->duplicate selected object) or Std_Copy (Edit->copy) on PartDesign body(ies) or feature(s), the Object Selection dialog box has the selected objects as well as dependencies all checked. In my opinion, you almost never want to copy dependencies. In fact, copying dependencies which are features of another body (e.g. a sketch is created from a master sketch in another body by Sketcher_Carbon_Copy), this results in PD features outside of any body.

Manually going through the object list to uncheck objects not specifically selected before starting the operation is tedious and error prone. Conversely, if only the original objects are automatically checked, it is easy to go through and check everything else.

Assuming that we don't want to change the current default behavior (even though I personally think the current behavior is most often wrong), I'd suggest a button in the dialog box to "uncheck dependencies" and possibly a button "check dependencies".

Also useful would be a configuration parameter to set the default behavior.

# Steps to reproduce

1. Open new document and switch to Part Design WB.
2. Create a body and sketch with a circle in the center and close the sketch.
3. Pad the sketch.
4. Make only the sketch visible.
5. Create a second body and sketch.
6. Make sure construction lines is toggled off, click Carbon Copy and control-click the visible sketch. Close the sketch.
7. Pad this sketch.
8. Double-click the second body to deactivate it and single-click to select that body.
9. Edit->duplicate selection. The select objects dialog box has, in addition to the contents of Body001, XY_Plane and Sketch from the original body. Click OK.
10. We now have Body002 which is a copy of Body001 EXCEPT that Sketch001 references Sketch002 instead of Sketch. Also Sketch002 (a copy of Sketch) and XY_Plane002 are floating outside of any body.

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **Updated:** 6/6/2021

# Discussion from Mantis ticket

### Comment by TedM 2021-06-05 06:12
Forum thread: [https://forum.freecadweb.org/viewtopic.php?f=3&t=59098](https://forum.freecadweb.org/viewtopic.php?f=3&t=59098)

---

### Comment by TedM 2021-06-05 06:13
I am currently unable to upload my sample file. I'm getting the following:

APPLICATION ERROR #503

Invalid upload path. Directory either does not exist or not writable to webserver.

Please use the "Back" button in your web browser to return to the previous page. There you can correct whatever problems were identified in this error or select another action. You can also click an option from the menu bar to go directly to a new section.

---

### Comment by TedM 2021-06-06 01:24
I have placed the sample file in the 10th message of the linked thread. The file earlier in the thread will also work, it's just a tiny bit more complex.

---

### Comment by Pauvres_honteux 2021-06-06 06:37
Hmm, to me this sounds like a version of [4502: Automate Link and ShapeBinder](https://tracker.freecadweb.org/view.php?id=4502)

I.e. just copy, then decide for a pasting option: with link, without link (as is) or dead.

adrianinsaval, 2022-02-10:
A feature to unselect dependencies and copy only original selection has been implemented, this can be closed IMO

chennes, 2022-02-10:
@adrianinsaval  Do you know when that was added, or in what commit? I'd like to have a backlink here for future reference.

adrianinsaval, 2022-02-10:
#5010

closed 2022-02-10 by chennes