FreeCAD/FreeCAD #6233: Recomputing after unchanged task/sketch

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4688

* **Reporter:** hgraminho
* **Date submitted:** 6/23/2021
* **FreeCAD version:** 0.19
* **Category:** Bug
* **Status:** new
* **Tags:** 

# Original report text

There are certain feature types for which opening their respective task panels for edit, followed by leaving them unchanged (either by pressing cancel or esc), causes them to become marked to recompute. This causes unnecessary delays, especially for larger models.

Comparing to version 0.18, this behavior was fixed for many of the tools, but also appeared on others, including the sketcher.

As far as I tested (mostly PartDesign workbench), leaving or cancelling tool task pannel...
...generates an undesired "mark to recompute" for:
FC 0.18: additive and subtractive pipes, mirrored feature, linear pattern and polar pattern
FC 0.19: sketches, additive and subtractive pipes

...does not generate "mark to recompute" (which is fine*) for:
FC 0.18: sketches, pad, pocket, lofts, grooves, chanfers, fillets, thickness
FC 0.19: pad, pocket, lofts, grooves, chanfers, fillets, thickness, mirrored feature, linear pattern and polar pattern

*however, I noticed that if there is some editable field that gets changed, then even if you change the value back and/or cancel the tool, "mark to recompute" will be generated as well. This seems to be happening for Part WB too.

Could be happening for other workbenches as well.

# Additional information

Forum discussion (behavior confirmed by other):
https://forum.freecadweb.org/viewtopic.php?f=8&t=58588&p=503530#p503530

Since this fix possibly requires changes to multiple tools and workbenches, I suggest fixing the Sketcher first, since it plays a central role for most of the users.

# Steps to reproduce

Create partdesign body.
Create 2 simple sketches, a closed one and a path (e.g. straight line).
Create additive pipe.
Activate "Skip recomputes"
Open pipe task panel (double click on it in the tree)
Press cancel or esc.
Note the mark for recompute icon.
Press F5 (update)
Enter and leave either sketch without changing it.
Note the mark for recompute icon.
Press F5 (update)

Now, just for comparison, try creating a pad or any of the other features I mentioned were ok.
Hit F5
Open feature task panel, then cancel. Notice that mark for recompute is NOT generated.
Now do the same but try editing something within the panel before cancel. Recompute IS generated.

# FreeCAD Info

```
OS: Windows 10 Version 2004
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24291 (Git)
Build type: Release
Branch: releases/FreeCAD-0-19
Hash: 7b5e18a0759de778b74d3a5c17eba9cb815035ac
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: Portuguese/Brazil (pt_BR)

```

# Other bug information

* **Priority:** normal
* **Severity:** tweak
* **Category:** Bug
* **Updated:** 6/23/2021

luzpaz, 2023-11-13:
Bumping for visibility

closed 2024-12-29 by github-actions[bot]

NXTler, 2024-12-29:
This issue still persist as far as I know.