FreeCAD/FreeCAD #6234: Sketch partially vanishes on zoom in

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4689

* **Reporter:** chrisb
* **Date submitted:** 6/27/2021
* **FreeCAD version:** 0.2
* **Category:** Bug
* **Status:** new
* **Tags:** 

# Original report text

In Sketcher parts of the still visible area vanish. Guess from Abdullah: Coin miscalculates and assumes that some parts are outside of the visible area which in fact aren't.
[Forum post](https://forum.freecadweb.org/viewtopic.php?f=3&t=59680).

# FreeCAD Info

```
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.25131 (Git)
Build type: Release
Branch: master
Hash: 7c519689f0d5ea78fb3292be36a857d283c05507
Python version: 3.9.5
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.2
Locale: C/Default (C)

```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **Updated:** 6/27/2021

kisolre, 2022-04-28:
When MapReverse is On the elliptical sector (Pocket.Edge17) is somehow wrongly mapped to the sketch plane and the weirdness starts. Delete the external edge - no issue. Add it again (not undo) - same weirdness
[image]
.

adrianinsaval, 2023-04-21:
this seems a duplicate of #6836

Jolbas, 2023-06-01:
I don't think it's related to #6836. I had a similar issue when a transform matrix got invalid making some of the placement properties contain NaN values. Which then made a mess of the objects bounding boxes. Coin  seemed to get confused by that and no longer knows where to put the clipping planes. At least the clipping behaves in a similar manner. Maybe related and there is some erroneous part in sketch that spreads.

maxwxyz, 2024-02-25:
I cannot reproduce, also the demo file in the forum post seems to work.
Can anyone confirm, otherwise I'll close this.
<details><summary>About Info</summary>
<p>

```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36082 (Git)
Build type: Release
Branch: main
Hash: 7a025e4e60a607bdbdd1599627ce13bb8eeda071
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.2
  * CurvedShapes 1.0.5
  * Curves 0.6.23
  * Defeaturing 1.2.2
  * fasteners 0.5.12
  * FEMbyGEN 2.1.0
  * freecad.gears 1.0.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.4.2
```

</p>
</details>

luzpaz, 2024-02-26:
@maxwxyz can you ping the forum thread related to the ticket ?

Jolbas, 2024-02-26:
I can't reproduce any longer on MacOS either. It's there in 0.21 but not in 0.22.

adrianinsaval, 2024-03-06:
likely fixed by #12306

closed 2024-03-06 by maxwxyz