FreeCAD/FreeCAD #6236: [Feature]  'Hard' Block Constraint / Non-Editable Edges

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4695

* **Reporter:** paullee
* **Date submitted:** 7/2/2021
* **FreeCAD version:** 0.2
* **Category:** Feature
* **Status:** assigned
* **Tags:** sketcher

# Original report text

[Post-new-feature-cleanup] - [Chrisb's comment](https://forum.freecadweb.org/viewtopic.php?f=9&t=26572&p=514021#p514011)

FreeCAD Forum Discussion:-
1. [ Edges with Block Constraints still Moves](https://forum.freecadweb.org/viewtopic.php?f=3&t=54004&start=10#p481842)
2. [Block Constraint](https://forum.freecadweb.org/viewtopic.php?f=9&t=26572&start=30#p273230)

***Desirable 'Hard' Block Constraint / Non-editable Edge Behaviour***

Excerpt from above forum discussion thread :

"Now, it seems there are 2 'intended behaviour' needs to be implemented, one is 'hard' - the previous behaviour as depicted in wiki, and the new 'soft' one :D

I use extensively the Block Constraint in my model and in the SketchArch add-on, assuming the edge with Block Constraint would never move again :) The new user in the post I refer seems expected similar behaviour. My usescases e.g.:-

    1. The X, Y Axis and Origin never move
    2. Sometime edges in other placement / direction are needed as reference (we have Grid in architectural design)
    3. In below model, there are a few edges that are originally added making reference to edges of a Ramp
    4. They are Blocked and worked as 'External Geometry', not expected to be Driven - and Not Susceptible to 'Toponaming' problem
    5. Then I build few edges / geometry based on these reference line, those edges / geometry are driven by the Blocked Edges
    6. The thing now it become 'non-intuitive' to those used to old behaviour is that - when there are more dimensional added to the latter edges, then suddenly it would come to a moment, a turning point, that the Blocked Edge become Driven by those originally Be Driven in opposite way. (Try to illustrate in below screencaptures... the last one capture the Block Edge 'suddenly' move / 'be driven'.

Can a 'Hard Block' / 'Absolute Block' constraint be re-provided in addition to the one with new behaviour ? :D "

***'Soft' Block Constraint Behaviour Introduced in Dec 2020***

abdullah :
" This is, as far as I see, the intended behaviour of the new Block constraint (introduced in December 2020).

The working of new Block can be summarised as: "Block on a geometry constraints anything else not being constraint by another constraint".

The old block constraint led to misleading behaviour (sometimes a dimension would be ignored while still showing the new value) and inconsistent behaviour (whether the constraint is ignored or shown as redundant/conflicting depends on the position of the dimensional constraint with respect to the block constraint in the list of constraints).

The new block always obeys any other existing constraint (no matter whether it is located before or after in the constraint list) and fixes the rest. It is my understanding that this is the wanted behaviour."

# Steps to reproduce

(see above / forum discussion with links provided)

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 7/2/2021

# Discussion from Mantis ticket

### Comment by paullee 2021-07-02 08:36
Add [Post-new-feature-cleanup] following [Chrisb's comment](https://forum.freecadweb.org/viewtopic.php?f=9&t=26572&p=514021#p514011)

paullee0, 2025-03-22:
Not sure of the way forward particularly after abdullah's absence?  :)

Roy-043, 2026-11-14:
Implemented with #22217.

closed 2026-11-14 by Roy-043