FreeCAD/FreeCAD #6237: Problems with startup from command line (attached logs)

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4696

* **Reporter:** mcgiwer
* **Date submitted:** 7/2/2021
* **FreeCAD version:** 0.19
* **Category:** Bug
* **Status:** assigned
* **Tags:** 0.19

# Original report text

When I attempt to start a fresh (just downloaded from the freecad.org website) FreeCAD from the console and add an " & " char at the end (to run it in background), I get a following error message thrown out in the terminal window:

```
/tmp/.mount_freecazyi6sl/AppRun: line 41: 21741 Aborted                 ${MAIN} "$@"
```

# Additional information

For security reasons, I had replaced in the attached debug log the path to the home folder

# Steps to reproduce

Attempt to start FreeCAD from the console, adding after the command a space and the & character

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **OS: Debian 10**
* **Platform:** amd64
* **Updated:** 1/25/2022

# Discussion from Mantis ticket

### Comment by mcgiwer 2021-07-14 18:28
[UPDATE]

I had discovered that the issue was caused because xcb couldn't become loaded. I got able to fix this by typing in the console following command:

`sudo xhost +si:localuser:$USER`

---

### Comment by mcgiwer 2021-07-18 19:30
[UPDATE]

***Note:*** The path where freeCAD is placed has been truncated for security reasons

After updating python to the version 3.9, it stopped to work with following errors:

`
Python path configuration:
  PYTHONHOME = '/usr/lib/python3.9' (set manually)
  PYTHONPATH = (not set)
  program name = 'bin/freecad'
  isolated = 0
  environment = 1
  user site = 1
  import site = 1
  sys._base_executable = 'bin/freecad'
  sys.base_prefix = '/usr'
  sys.base_exec_prefix = '/usr'
  sys.executable = 'bin/freecad'
  sys.prefix = '/usr'
  sys.exec_prefix = '/usr'
  sys.path = [
    '/usr/lib/python38.zip',
    '/usr/lib/python3.8',
    '/usr/lib/python3.8/lib-dynload',
  ]
Fatal Python error: init_fs_encoding: failed to get the Python codec of the filesystem encoding
Python runtime state: core initialized
ModuleNotFoundError: No module named 'encodings'

Current thread 0x00007fd8b4507800 (most recent call first):
<no Python frame>
`
Additionally, I don't know why the "PYTHONPATH" variable is displayed as not beeing set, even if it's has been exported.

***Enviromental informations:***

*  System: Debian 10 (64-bit Linux)
*  FreeCAD version: 0.19.2
*  Installed Python versions: 2.7 (for copatibility reasons to be able to run some extensions), 3.9

Hopefully it helps

---

### Comment by chennes 2021-08-11 17:20
It looks like this is two different bugs, and that the first one has been resolved -- is that correct?

---

### Comment by chennes 2021-12-29 03:26
mcgiwer Your original bug report appears to have been resolved and was related to your system configuration. My read is that your later post here is a totally separate bug, so I plan to close this one shortly. If that is not the case, please let me know.

---

### Comment by Kunda1 2022-01-25 18:17
mcgiwer ping

---

### Comment by mcgiwer 2022-01-25 18:23
Kunda1 pong ;-p

---

### Comment by mcgiwer 2022-01-25 18:26
chennes The both bugs are related to each other and I'm not sure why they are becomming thrown out. To add it, this issue has appeared suddenly because in not long time earlier, the FreeCAD was working correcly. I hope it's only a temporary problem

adrianinsaval, 2022-02-10:
forum discussion: https://forum.freecadweb.org/viewtopic.php?f=4&t=60321

This should be closed, xcb issue seems to be specific to OP's system and not really related to FreeCAD as he reported other apps had the issue too.

The other error comes from absurdly incorrect use of the appimage so can be dissmissed.

chennes, 2022-02-10:
Thanks for the feedback, @adrianinsaval -- closing per your advice. To the original poster: you are welcome to continue discussing the situation in the forums topic linked above.

closed 2022-02-10 by chennes