FreeCAD/FreeCAD #6238: Pattern not working properly on geometric primitives

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4698

* **Reporter:** chrisb
* **Date submitted:** 7/3/2021
* **FreeCAD version:** 0.2
* **Category:** Bug
* **Status:** new
* **Tags:** 

# Original report text

In some cases the PartDesign pattern functions don't seem to get the coordinate system of geometric primitives right. The patterned features occur then in the wrong place.
[forum discussion](https://forum.freecadweb.org/viewtopic.php?f=3&t=57658)

# FreeCAD Info

```
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.25157 (Git)
Build type: Release
Branch: master
Hash: 257cdd8f423f1530b8df6f53e242de629fee4e85
Python version: 3.9.5
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.2
Locale: C/Default (C)

```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **Updated:** 7/26/2021

# Discussion from Mantis ticket

### Comment by openBrain 2021-07-26 16:22
Just notice this is one of all things that don't work with PD primitives : mirror, ...
Indeed it seems that primitives are by some way messing up the coordinate system.

adrianinsaval, 2023-03-10:
@realthunder given your recent PR about PD patterns, could you also take a look into this one? or has this been solved by that PR too?

FEA-eng, 2024-10-08:
This still seems to be the case even after #16963 was merged.

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38923 (Git)
Build type: Release
Branch: main
Hash: d20cb9e6ee198beb2bfd7e72d3dec0a575e3f28c
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

FlachyJoe, 2024-10-08:
@FEA-eng Yes, primitives placement still needs work. See also https://github.com/FreeCAD/FreeCAD/issues/6032#issuecomment-2397737306

FlachyJoe, 2025-01-07:
Still relevant.