FreeCAD/FreeCAD #6239: Add the possibility to dimension two parallel lines

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4700

* **Reporter:** hic
* **Date submitted:** 7/3/2021
* **FreeCAD version:** 0.2
* **Category:** Feature
* **Status:** new
* **Tags:** 

# Original report text

Make it possible to set the distance between two parallel lines. 

At the moment it is possible to set the distance between a point and a line or between two points. But it is impossible to set the distance between two lines.

Additionally, automatically generate a parallelism constraint between these lines. Maybe add a warning if they are not parallel at this point.

# Additional information

Link to forum discussion:

[https://forum.freecadweb.org/viewtopic.php?f=3&t=54945](https://forum.freecadweb.org/viewtopic.php?f=3&t=54945)

# FreeCAD Info

```
OS: Linux Mint 19 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.20.25157 (Git) AppImage
Build type: Release
Branch: master
Hash: 257cdd8f423f1530b8df6f53e242de629fee4e85
Python version: 3.9.5
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.2

```

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 7/3/2021

Zolko-123, 2022-10-25:
It's trivial to do without, this should be closed

luzpaz, 2022-10-30:
@paddlestroke 
re: https://forum.freecadweb.org/viewtopic.php?p=598547#p598547
#5407

PaddleStroke, 2022-10-30:
Yes, constrain contextually is still pending review.

luzpaz, 2022-10-31:
@PaddleStroke this feature depends on #5407 and can be closed when merged ?

PaddleStroke, 2022-10-31:
Well I'm not touching the distance constraint  tool itself. Though in my opinion constrain contextually should make distance, distanceX, distanceY, Angle and radius/diameter totally obsolete. If you ask me those tools should be removed when constrain contextually merges.
But knowing the forum this is not going to happen. So I'm not sure what to say.

adrianinsaval, 2023-08-30:
if the lines are already parallel that is true, however it would be nice if the could allow making the lines parallel if they aren't exactly parallel yet. Currently it offers angle and equality with the shift key, parallelism might be worth adding as option when cycling with shift key

howie-j, 2023-08-30:
True. Also multiple points shoud give the option of coincident. Perhaps we should open another issue and discuss Sketcher_Dimension alternate constraints there?

PaddleStroke, 2023-08-30:
In my initial implementation there was those modes indeed. I simplified the code to make it simpler so that we could get it to merge.
Feel free to open issues and I'll implement them one by one.

adrianinsaval, 2023-08-30:
> True. Also multiple points shoud give the option of coincident. Perhaps we should open another issue and discuss Sketcher_Dimension alternate constraints there?

yes please do so, I'll close this issue since the basic desired functionality is there already

pierreporte, 2023-12-05:
@adrianinsaval @luzpaz Can we reopen this? The lack of real line to line distance makes the extension line cover the lines entirely, masking their type. @PaddleStroke wrote that this is the core problem of issue #11603.

luzpaz, 2023-12-06:
Reopened

pierreporte, 2024-04-12:
@PaddleStroke What about this?

FlachyJoe, 2024-04-12:
IMHO Sketcher currently propose "simple" constraints if we open the door of "complex" ones (multiple constraints set by one tool) then tool count will explode quickly.

pierreporte, 2024-04-12:
@FlachyJoe You are right. My question was about the core issue: distance between lines.

FlachyJoe, 2024-04-12:
Distance between lines is not an issue as it is already doable with end point to line.

pierreporte, 2024-04-12:
Still, it causes #11603.

FlachyJoe, 2024-04-12:
It's not the lack of this feature which causes the pointed problem but the constraint's line end calculation.

YaphetS1, 2024-09-03:
Maybe someone knows a plugin for this? Since the feature still hasn't been added?

FlachyJoe, 2024-09-03:
@YaphetS1 As already said you can constraint distance between two lines with Parallel and PointToLine distance constraint. Nothing more is needed.

[image]