FreeCAD/FreeCAD #6240: Object snap in sketcher

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4702

* **Reporter:** NewJoker
* **Date submitted:** 7/5/2021
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** new
* **Tags:** midpoint, snap

# Original report text

For me the most missing feature in FreeCAD is the lack of auto snapping to objects (midpoints, center points, quadrants, extensions, intersections) in sketcher.

Forum thread: https://forum.freecadweb.org/viewtopic.php?f=8&t=59890

# FreeCAD Info

```
OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.22808 (Git)
Build type: Release
Branch: master
Hash: 72eb41b24f12b572d55081042160954b93f4614c
Python version: 3.6.8
Qt version: 5.12.1
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: Polish/Poland (pl_PL)
```

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 7/5/2021

# Discussion from Mantis ticket

### Comment by NewJoker 2021-07-05 21:51
Forum thread started by another user: https://forum.freecadweb.org/viewtopic.php?f=8&t=58207

reynolga, 2023-01-26:
I was just searching for this feature too..

luzpaz, 2023-03-08:
@FEA-eng see #8387

adrianinsaval, 2023-04-15:
should we close this issue now that there is snapping in sketcher? is something still missing in regards to that?

FEA-eng, 2023-04-15:
@adrianinsaval Nope, it's not what is requested here. The new snap feature just adds snapping at an angle and introduces the snap manager. Snapping to midpoints, center points, quadrants, extensions or intersections is still missing.

adrianinsaval, 2023-04-15:
Are you sure you have tested latest master? I get at least snapping to midpoints, to grid (is that what you mean with quadrants?), to points (this includes center points) and to edges.

I think you need to update this issue to be specifically about what is actually missing, otherwise it's not useful.

About snapping to extensions, I don't think that's desirable...

About intersections I guess that can be considered related to #6026 although I think that one was more about auto constraints.

adrianinsaval, 2023-04-15:
Inviting @PaddleStroke

FEA-eng, 2023-04-15:
@adrianinsaval Yes, I tested this in the newest Windows build (32821). By midpoints, I mean midpoints of lines and arcs. That's not implemented yet but paddle mentioned adding it in the future. This one would be the most important for me but other snaps would be useful too. Quadrant snap would snap to the quadrant points on the circumference of circles, arcs and ellipses. Snapping to extensions is available in other CAD software and can be very useful but would probably require more effort to implement.

adrianinsaval, 2023-04-15:
Midpoints work for me, are you confusing snaps and auto constraints?

[image]

can you describe common usecases for quadrant snaps? on first thought it seems very niche to me.

FEA-eng, 2023-04-16:
@adrianinsaval Right, I haven't noticed that midpoint snap working before. It's really great to have it already available. What would be needed to make it equivalent to snaps in other CAD software is an auto constraint and maybe some symbol indicating a successful snap.

Quadrant snap can also be quite useful. For example, to properly draw this (btw. tangent snap would also be nice):

[image]

When it comes to the previously mentioned (geometric) center snap, it would be useful for closed profiles made of lines/polylines.

PaddleStroke, 2023-04-17:
Currently midpoints are only snapping, there are no auto-constraint generated.
For lines we could have a symmetric constraint. But it needs to be implemented.
For arcs I'm not sure what we could use.

FEA-eng, 2024-04-01:
> Maybe issues for specific object snapping that are missing can be opened (and are potentially already opened)

Done:
- https://github.com/FreeCAD/FreeCAD/issues/6026
- https://github.com/FreeCAD/FreeCAD/issues/13266
- https://github.com/FreeCAD/FreeCAD/issues/13267
- https://github.com/FreeCAD/FreeCAD/issues/13268
- https://github.com/FreeCAD/FreeCAD/issues/13269

maxwxyz, 2024-04-02:
I'll close this one then.

closed 2024-04-02 by maxwxyz