FreeCAD/FreeCAD #6242: [Draft] toolbar is active for all workbenches

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4705

* **Reporter:** uwestoehr
* **Date submitted:** 7/16/2021
* **FreeCAD version:** 0.19
* **Category:** Bug
* **Status:** assigned
* **Tags:** 

# Original report text

- change to the Draft toolbar
- close FreeCAD
- open FreeCAD

result: no matter in what workbench you are, the toolbar containing Draft's "Create working plane proxy" is always shown, and moreover active. As it is active, you can click on the toolbar buttons and get errors.

The bug is in FC 0.19. and current FC 0.20 master.

# FreeCAD Info

```
Forum thread: https://forum.freecadweb.org/viewtopic.php?f=3&t=60215

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24291 (Git)
Build type: Release
Branch: releases/FreeCAD-0-19
Hash: 7b5e18a0759de778b74d3a5c17eba9cb815035ac
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: German/Germany (de_DE)
```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **Updated:** 12/16/2021

# Discussion from Mantis ticket

### Comment by uwestoehr 2021-07-16 12:09
In the forum this workaround was reported:
- open FC
- go to the Draft WB
- go to another WB

then the toolbar disappears.

---

### Comment by yorik 2021-12-16 15:17
[quote=uwestoehr]- change to the Draft toolbar
- close FreeCAD
- open FreeCAD

result: no matter in what workbench you are, the toolbar containing Draft's "Create working plane proxy" is always shown, and moreover active. As it is active, you can click on the toolbar buttons and get errors.[/quote]

I cannot reproduce this (master branch). The bar is not present on relaunching FreCAD. Can you look again uwestoehr ?

---

### Comment by uwestoehr 2021-12-16 22:17
> I cannot reproduce this (master branch).

This is still there (one of my favorite "annoyants" ;-) )

1. open FC
2. set in the preferences that FC is started with the PartDesign workbench
3. change to the Draft WB
4. close FC and restart it

result: the Draft toolbar is there, see attached image

OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.26656 (Git)
Build type: Release
Branch: master
Hash: 8585149ab4ee0c165c5c443e84915741968d5624
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.3
Locale: German/Germany (de_DE)

adrianinsaval, 2022-02-11:
can reproduce here:
```
OS: Linux 5.16.8-arch1-1 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.27319 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 67706b1)
Hash: 67706b14df32b91b948952ea2c9350948ba4d4db
Python version: 3.9.10
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.3
Locale: English/United Kingdom (en_GB)
```

but not here:
```
OS: Linux 5.16.8-arch1-1 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.27465 (Git)
Build type: Release
Branch: master
Hash: 092afacfcb5127eb67c385a43cbe34a4cd15bef4
Python version: 3.10.2
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United Kingdom (en_GB)

```
it's a Conda specific issue

luzpaz, 2022-02-11:
@looooo should we transfer this ticket to the FreeCAD-bundle repo ?

adrianinsaval, 2022-02-11:
Sorry! spoke too soon, I can also reproduce in both versions so it's not conda specific

luzpaz, 2022-03-19:
I can reproduce this on the Snap Edge 

1. start FC
2. switch to Draft wb
3. close FC
4. start FC
Result:
[image]

[code]
OS: Ubuntu Core 20 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.28163 (Git)
Build type: Release
Branch: master
Hash: 88daed86eab75dc85a46c11e670b4a55cc2cb36a
Python 3.8.10, Qt 5.15.3, Coin 4.0.0, OCC 7.6.1
Locale: C/Default (C)
Installed mods: 
  * ToolbarStyle
  * Assembly4 0.11.10
  * stickframe
  * sheetmetal 0.2.49
  * Behave-Dark-Colors 0.0.1
  * Curves 0.3.0
  * Ship
  * BIM 2021.12.0
  * A2plus 0.4.54b
[/code]

donovaly, 2022-03-19:
I added it to the 0.20 milestone sine I guess Pythionists like @sliptonic or @yorikvanhavre know a fix.

yorikvanhavre, 2022-04-06:
On it!

yorikvanhavre, 2022-04-08:
Damn this is complicated.. I can't find what makes this widget show. It seems in any case bound to the qt objectName property. If one changes this objectName ( https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/Draft/DraftGui.py#L302 ), on next FreeCAD run, things works as expected. On subsequent run, the new name is "remembered" somewhere, and the bar is shown on next start. I think there is a mechanism that remembers which toolbar should be shown or hidden on next run (that exists in the preferences but no effect), but that somehow fails in this case.

commit 9848d601 (2022-05-09): [BugFix] Draft: ensure Working Panel toolbar is deactivated on FC closing

 Fixes #6242
closed 2022-05-09 by yorikvanhavre